espressif/esp-dsp #2: 如何使用xTaskCreatePinnedToCore函数来加快esp32的运算速度

Herwey, 2019-03-19:
Hi:
我现在使用esp32进行图像处理，使用xTaskCreatePinnedToCore( imageprocess, " imageprocess", 1024*10, NULL, 10, NULL, 0 ) 来设定图像处理函数指向0核。但是发现使用这个函数把任务指向0核和1核，两者的运算速度并没有提升很多。请问一下这是什么原因？使用的是C语言。谢谢！

dmitry1945, 2019-03-19:
Hi Herwey, 
If I understand your question correct, you trying to use xTaskCreatePinnedToCore to increase a speed of processing.
If you have WiFi or Bluetooth task that handles the Wi-Fi/BT, it's useful to allocate one CPU core to WiFi and second core to your application. In this case WiFi will not interrupt your task. Also all operation system tasks will run completely on one core and will not interrupt your task. 
It depends on the application, but I think in case of image processing there is no way to increase the speed by assigning dedicated core.
The Esp32 has two cores. You can try to split your task - "imageprocessing" into two parallel tasks that will do independent work in parallel, and assign them to two CPU cores by xTaskCreatePinnedToCore. In this case you will increase performance.

Regards,
Dmitry

Herwey, 2019-03-19:
Hi Dmitry
I found here was a simple misunderstanding, I used ESP32-s, that means it only has single core. So...emmm, thanks any way!
Best Regards !

hughhugh, 2019-03-21:
@dmitry1945 
Hi, how can I use the  esp32-DSP library to speed up the image processing when I using **dl_matrix3du_t**  to allocate memory for the image data. The algorithm I use is a two-pass method. It is so slow.  Do you have some suggestions?
Thank you.

dmitry1945, 2019-03-21:
@hughhug
Hi,
I don't know the details of your project, that's why I can make only basic suggestions. 
First, identify where you loose your performance. It could be memory/cache, or it could be the function itself.
If it's memory, then try move your processing data to the internal RAM first (partly), and then process them. Organize it as pipeline.
If it's a function itself, you can find similar function in esp-dsp and change them, or you can rewrite your function in assembler. The assembler is about 2..3 times faster. In esp-dsp you will find a lot of functions with assembler. You can look how to do this.

Regards,
Dmitry

Herwey, 2019-03-21:
@dmitry1945
Hi,
But how to process data to the internal RAM ? Can you provide some reference or demos ? 
Thanks in advance !

dmitry1945, 2019-03-22:
@hughhug
Hi,
In esp-idf/examples/peripherals/timer_group/main/timer_group_example_main.c
you will see function:
```
void IRAM_ATTR timer_group0_isr(void *para)
{
...
}

```
Attribute IRAM_ATTR means that source code for the function will be placed into the IRAM (internal instruction RAM).

If you use arrays with constants, like:
const int some_data[] = {0,1,2,3,4};
Try to allocate a buffer, and copy some_data to this buffer. Or just remove "const".
If you use PSRAM, then first copy data to the buffer, and then process it.
Regards,
Dmitry

closed 2023-07-28 by dmitry1945