espressif/esp-dsp #4: Explain FFT example

tatulea, 2019-07-08:
Hi, I am looking to do FFT on a signal coming from the accelerometer.
I ahve 512 sample readings and I know the frequency. I used FFTW library in python to do the FFT for the signal and it's working.
How can I use this library to get the same thing? I had a look over your example, but I don't understand exactly what x1 and x2 are and how I should modify it.

Regards,
Gabriel

dmitry1945, 2019-07-08:
Hi Gabriel,
The esp-dsp library working with complex numbers, that's why the simplest way for you will be just ignore the x2 input and y2 output, and use only x1 and y1. Your code will look so:

```
    esp_err_t ret;
    ESP_LOGI(TAG, "Start Example.");
    ret = dsps_fft2r_init_fc32(NULL, CONFIG_DSP_MAX_FFT_SIZE);
    if (ret  != ESP_OK)
    {
        ESP_LOGE(TAG, "Not possible to initialize FFT. Error = %i", ret);
        return;
    }
   // Generate hann window
    dsps_wind_hann_f32(wind, N);
    // Generate input signal for x1 A=1 , F=0.1
    dsps_tone_gen_f32(x1, N, 1.0, 0.16,  0);
    // x1 - is your signal that you have to fill
    // Convert two input vectors to one complex vector
    for (int i=0 ; i< N ; i++)
    {
        y_cf[i*2 + 0] = x1[i] * wind[i]; // Real part is your signal multiply with window
        y_cf[i*2 + 1] = 0; // Imag part is 0
    }
    dsps_fft2r_fc32(y_cf, N);
    unsigned int end_b = xthal_get_ccount();
    // Bit reverse 
    dsps_bit_rev_fc32(y_cf, N);
    // Convert one complex vector to two complex vectors
    dsps_cplx2reC_fc32(y_cf, N);
    
    // y1_cf - is your result in log scale
    // y2_cf - magnitude of your signal in linear scale

    for (int i = 0 ; i < N/2 ; i++) {
        y1_cf[i] = 10 * log10f((y1_cf[i * 2 + 0] * y1_cf[i * 2 + 0] + y1_cf[i * 2 + 1] * y1_cf[i * 2 + 1])/N);
        y2_cf[i] = ((y1_cf[i * 2 + 0] * y1_cf[i * 2 + 0] + y1_cf[i * 2 + 1] * y1_cf[i * 2 + 1])/N);
    }
  
    // Show power spectrum in 64x10 window from -60 to 0 dB from 0..N/2 samples
    ESP_LOGW(TAG, "Signal x1 in log scale");
    dsps_view(y1_cf, N/2, 64, 10,  -60, 40, '|');
    ESP_LOGW(TAG, "Signal x1 in absolute scale");
    dsps_view(y2_cf, N/2, 64, 10,  0, 2, '|');

```

tatulea, 2019-07-09:
Hi Dmitry,

I am using this xlsx file as input: [512.xlsx](https://github.com/espressif/esp-dsp/files/3371976/512.xlsx)
It has 512 values and the FFT result that I get using the python script is this: [fft_output.xlsx](https://github.com/espressif/esp-dsp/files/3371975/fft_output.xlsx)

How can I use you example to get the same result?

dmitry1945, 2019-07-09:
Can you show me your python script?

tatulea, 2019-07-09:
The code that I am using with python is from mide.com

It is not too clean and I've modified few things to accommodate my data input.

```
import matplotlib.pyplot as plt
import pandas as pd
import numpy as np
import tkinter as tk
from tkinter import filedialog
import time
import pyfftw
import os

#####
# Functions for testing various FFT methods
#####

def fftw_builder_test(input_data):
    pyfftw.forget_wisdom()  # This is just here to keep the tests honest, normally pyfftw will remember setup parameters and go more quickly when it is run a second time
    a = np.array(input_data,
                 dtype='float32')  # This turns the input list into a numpy array. We can make it a 32 bit float because the data is all real, no imaginary component
    fft_obj = pyfftw.builders.rfft(a)  # This creates an object which generates the FFT
    return fft_obj()  # And calling the object returns the FFT

def fftw_fast_builder_test(input_data):  # See fftw_builder_test for comments
    pyfftw.forget_wisdom()
    a = np.array(input_data, dtype='float32')
    fft_obj = pyfftw.builders.rfft(a,
                                   planner_effort='FFTW_ESTIMATE')  # FFTW_ESTIMATE is a lower effort planner than the default. This seems to work more quickly for over 8000 points
    return fft_obj()

def fftw_test(input_data):
    pyfftw.forget_wisdom()  # This is just here to keep the tests honest
    outLength = len(
        input_data) // 2 + 1  # For a real FFT, the output is symetrical. fftw returns only half the data in this case
    a = pyfftw.empty_aligned(len(input_data),
                             dtype='float32')  # This is the input array. It will be cleared when the fft object is created
    outData = pyfftw.empty_aligned(outLength,
                                   dtype='complex64')  # This is the output array. Not that the size and type must be appropriate
    fft_obj = pyfftw.FFTW(a, outData, flags=('FFTW_ESTIMATE',),
                          planning_timelimit=1.0)  # NB: The flags tuple has a comma
    a[:] = np.array(input_data,
                    dtype='float32')  # We have to fill the array after fft_obj is created. NB: 'a[:] =' puts data into the existing array, 'a =' creates a new array
    return fft_obj()  # Calling the object returns the FFT of the data now in a. The result is also in outData

def fftw_test_complex(
        input_data):  # This is fftw_test but running a complex FFT as opposed to a real input FFT. See fftw_test for comments
    pyfftw.forget_wisdom()
    outLengthMod = len(input_data) // 2 + 1  # Size of expected return data
    outLength = len(input_data)  # For a complex input FFT, we get more data
    a = pyfftw.empty_aligned(len(input_data),
                             dtype='complex64')  # The FFTW determines the type of FFT by the type of input
    outData = pyfftw.empty_aligned(outLength, dtype='complex64')
    fft_obj = pyfftw.FFTW(a, outData, flags=('FFTW_ESTIMATE',), planning_timelimit=1.0)
    a[:] = np.array(input_data, dtype='complex64')
    return fft_obj()[0:outLengthMod]

def fftw_test_default(input_data):  # This is fftw_test with different FFTW options. See fftw_test for comments
    pyfftw.forget_wisdom()
    outLength = len(input_data) // 2 + 1
    a = pyfftw.empty_aligned(len(input_data), dtype='float32')
    outData = pyfftw.empty_aligned(outLength, dtype='complex64')
    fft_obj = pyfftw.FFTW(a, outData)
    a[:] = np.array(input_data, dtype='float32')
    return fft_obj()

def fftw_test_no_limit(input_data):  # This is fftw_test with different FFTW options. See fftw_test for comments
    pyfftw.forget_wisdom()
    outLength = len(input_data) // 2 + 1
[... 83 lines cut ...]
        if include_plots:
            plt.figure(figure_count)
            figure_count += 1
            plt.plot(xf, 2.0 / N * np.abs(fftData))  # [0:np.int(N/2)]
            plt.grid()
            plt.xlabel('Frequency (Hz)')
            plt.ylabel('Accel (g)')
            plt.title('FFT (%s) - %s' % (func_name, file_path))
        toc = time.clock()
        output_data["FFT Time"] = toc - tic
        print("FFT Time:", toc - tic)
        write_csv(out_path, output_headers, output_data)

    output_data = []
    for i, elem in enumerate(xf):
        output_data.append(str(elem) + "," + str(2.0 / N * np.abs(fftData[i])))
    with open('/home/gabi/iesite.csv', 'w') as f:
        f.write("\n".join(output_data))
if include_plots:
    print("Plotting...")
    plt.show()
print("Done!")
```

On lines 176, 177 I get the output to csv and I think that is what you are looking for

dmitry1945, 2019-07-09:
OK. Just take y2_cf output and not use window from example that I have included.
Like this:
``` 
    esp_err_t ret;
    ESP_LOGI(TAG, "Start Example.");
    ret = dsps_fft2r_init_fc32(NULL, CONFIG_DSP_MAX_FFT_SIZE);
    if (ret  != ESP_OK)
    {
        ESP_LOGE(TAG, "Not possible to initialize FFT. Error = %i", ret);
        return;
    }
   // Generate hann window
    // Generate input signal for x1 A=1 , F=0.1
    dsps_tone_gen_f32(x1, N, 1.0, 0.16,  0);
    // x1 - is your signal that you have to fill
    // Convert two input vectors to one complex vector
    for (int i=0 ; i< N ; i++)
    {
        y_cf[i*2 + 0] = x1[i]; // Real part is your signal
        y_cf[i*2 + 1] = 0; // Imag part is 0
    }
    dsps_fft2r_fc32(y_cf, N);
    unsigned int end_b = xthal_get_ccount();
    // Bit reverse 
    dsps_bit_rev_fc32(y_cf, N);
    // Convert one complex vector to two complex vectors
    dsps_cplx2reC_fc32(y_cf, N);
    
    
    // y2_cf - magnitude of your signal in linear scale

    for (int i = 0 ; i < N/2 ; i++) {
       y2_cf[i] = ((y_cf[i * 2 + 0] * y_cf[i * 2 + 0] + y_cf[i * 2 + 1] * y_cf[i * 2 + 1])/N);
    }
  
    // Show power spectrum in 64x10 window from -60 to 0 dB from 0..N/2 samples
    ESP_LOGW(TAG, "Signal x1 in absolute scale");
    dsps_view(y2_cf, N/2, 64, 10,  0, 2, '|');

```

tatulea, 2019-07-09:
I am printing y2_cf elements, but they don't look like the ones from xlsx file. Shouldn't they be the same?

dmitry1945, 2019-07-09:
They should. Can you show them here?

dmitry1945, 2019-07-09:
And, the last example is different then the firs.

tatulea, 2019-07-09:
The code that I am using is:

```
    static float x[512] = {1.6,1.3,1,0.6,0.3,0.1,0,0.1,0.2,0.5,0.8,1.2,1.5,1.8,2,2,2,1.8,1.5,1.1,0.8,0.5,0.2,0.1,0,0.1,0.4,0.7,1,1.4,1.7,1.9,2,2,1.9,1.6,1.3,0.9,0.6,0.3,0.1,0,0.1,0.3,0.5,0.9,1.2,1.5,1.8,2,2,1.9,1.7,1.4,1.1,0.7,0.4,0.2,0.1,0,0.2,0.4,0.7,1.1,1.4,1.7,1.9,2,2,1.8,1.6,1.2,0.9,0.5,0.3,0.1,0,0.1,0.3,0.6,0.9,1.3,1.6,1.8,2,2,1.9,1.7,1.4,1,0.7,0.4,0.2,0,0.1,0.2,0.4,0.8,1.1,1.5,1.7,1.9,2,2,1.8,1.5,1.2,0.8,0.5,0.2,0.1,0,0.1,0.3,0.6,1,1.3,1.6,1.9,2,2,1.9,1.6,1.3,1,0.6,0.3,0.1,0,0.1,0.2,0.5,0.8,1.2,1.5,1.8,2,2,1.9,1.7,1.5,1.1,0.8,0.5,0.2,0.1,0,0.1,0.4,0.7,1,1.4,1.7,1.9,2,2,1.8,1.6,1.3,0.9,0.6,0.3,0.1,0,0.1,0.3,0.5,0.9,1.2,1.6,1.8,2,2,1.9,1.7,1.4,1.1,0.7,0.4,0.2,0,0,0.2,0.4,0.7,1.1,1.4,1.7,1.9,2,2,1.8,1.6,1.2,0.9,0.5,0.3,0.1,0,0.1,0.3,0.6,0.9,1.3,1.6,1.8,2,2,1.9,1.7,1.4,1,0.7,0.4,0.1,0,0.1,0.2,0.5,0.8,1.1,1.5,1.8,1.9,2,2,1.8,1.5,1.2,0.8,0.5,0.2,0.1,0,0.1,0.3,0.6,1,1.3,1.6,1.9,2,2,1.9,1.6,1.3,1,0.6,0.3,0.1,0,0.1,0.2,0.5,0.8,1.2,1.5,1.8,2,2,1.9,1.7,1.5,1.1,0.8,0.4,0.2,0.1,0,0.1,0.4,0.7,1,1.4,1.7,1.9,2,2,1.8,1.6,1.3,0.9,0.6,0.3,0.1,0,0.1,0.3,0.5,0.9,1.2,1.6,1.8,2,2,1.9,1.7,1.4,1.1,0.7,0.4,0.2,0,0,0.2,0.4,0.7,1.1,1.4,1.7,1.9,2,2,1.8,1.5,1.2,0.9,0.5,0.3,0.1,0,0.1,0.3,0.6,0.9,1.3,1.6,1.9,2,2,1.9,1.7,1.3,1,0.7,0.4,0.1,0,0.1,0.2,0.5,0.8,1.1,1.5,1.8,2,2,2,1.8,1.5,1.2,0.8,0.5,0.2,0.1,0,0.1,0.3,0.6,1,1.4,1.7,1.9,2,2,1.9,1.6,1.3,0.9,0.6,0.3,0.1,0,0.1,0.2,0.5,0.8,1.2,1.5,1.8,2,2,1.9,1.7,1.4,1.1,0.7,0.4,0.2,0,0,0.2,0.4,0.7,1.1,1.4,1.7,1.9,2,2,1.8,1.6,1.2,0.9,0.6,0.3,0.1,0,0.1,0.3,0.6,0.9,1.3,1.6,1.8,2,2,1.9,1.7,1.4,1,0.7,0.4,0.2,0,0.1,0.2,0.4,0.8,1.1,1.5,1.7,1.9,2,2,1.8,1.5,1.2,0.8,0.5,0.2,0.1,0,0.1,0.3,0.6,1,1.3,1.6,1.9,2,2,1.9,1.6,1.3,1,0.6,0.3,0.1,0,1.7,1.9,2,2,1.8,1.6,1.3,0.9,0.6,0.3,0.1,0,0.1,0.3,0.5,0.9,1.2,1.6,1.8,2,2,1.9,1.7,1.4,1.1,0.7,0.4,0.2,0,0,0.2,0.4,0.7,1.1,1.4,1.7,1.9,2,2,1.8,1.5,1.2,0.9,0.5,0.3,0.1,0,0.1,0.3,0.6,0.9,1.3};

    esp_err_t ret;
    ESP_LOGI(TAG, "Start Example.");
    ret = dsps_fft2r_init_fc32(NULL, CONFIG_DSP_MAX_FFT_SIZE);
    if (ret  != ESP_OK)
    {
        ESP_LOGE(TAG, "Not possible to initialize FFT. Error = %i", ret);
        return;
    }
   // Generate hann window
    // Generate input signal for x1 A=1 , F=0.1
    dsps_tone_gen_f32(x, N, 1.0, 0.16,  0);
    // x1 - is your signal that you have to fill
    // Convert two input vectors to one complex vector
    for (int i=0 ; i< N ; i++)
    {
        y_cf[i*2 + 0] = x[i]; // Real part is your signal
        y_cf[i*2 + 1] = 0; // Imag part is 0
    }
    dsps_fft2r_fc32(y_cf, N);
    unsigned int end_b = xthal_get_ccount();
    // Bit reverse 
    dsps_bit_rev_fc32(y_cf, N);
    // Convert one complex vector to two complex vectors
    dsps_cplx2reC_fc32(y_cf, N);
    
    
    // y2_cf - magnitude of your signal in linear scale

    for (int i = 0 ; i < N/2 ; i++) {
       y2_cf[i] = ((y_cf[i * 2 + 0] * y_cf[i * 2 + 0] + y_cf[i * 2 + 1] * y_cf[i * 2 + 1])/N);
       printf("%f\r\n", y2_cf[i]);
    }
  
    // Show power spectrum in 64x10 window from -60 to 0 dB from 0..N/2 samples
    ESP_LOGW(TAG, "Signal x1 in absolute scale");
    dsps_view(y2_cf, N/2, 64, 10,  0, 2, '|');
```

The output that I am getting is:

```
0.000244
0.000976
0.000977
0.000979
0.000982
0.000986
0.000991
0.000996
0.001003
0.001010
0.001019
0.001028
0.001038
0.001049
0.001062
0.001075
0.001090
0.001106
0.001123
0.001141
0.001161
0.001182
0.001205
0.001229
[... 222 lines cut ...]
0.000337
0.000336
0.000336
0.000336
0.000335
0.000335
0.000335
0.000335
0.000334
0.000334
W (558) main: Signal x1 in absolute scale
I (568) view: Data min[0] = 0.000244, Data max[82] = 501.346039
 ________________________________________________________________
0                    |                                           |
1                    |                                           |
2                    |                                           |
3                    |                                           |
4                    |                                           |
5                    ||                                          |
6                     |                                          |
7                   | |                                          |
8||||||||||||||||||||  |||||||||||||||||||||||||||||||||||||||||||
9                                                                |
 0123456789012345678901234567890123456789012345678901234567890123
```

dmitry1945, 2019-07-09:
Yes! :)
y2_cf[i] = sqrt((y_cf[i * 2 + 0] * y_cf[i * 2 + 0] + y_cf[i * 2 + 1] * y_cf[i * 2 + 1])/N);

))))

tatulea, 2019-07-09:
Is this working for you? The first values from my expected output are
```

x | y
-- | --
6.50976563 | 0.01152959
13.0195313 | 0.01720576
19.5292969 | 0.0210918
26.0390625 | 0.02718033
32.5488281 | 0.0254768
39.0585938 | 0.02137978
45.5683594 | 0.01765214
52.078125 | 0.01952043
58.5878906 | 0.00869537
65.0976563 | 0.01339019
71.6074219 | 0.0224928
78.1171875 | 0.02958141
```

And the values that I get from ESP are
```
0.000690
0.001381
0.001382
0.001383
0.001385
0.001388
0.001391
0.001395
0.001400
0.001405
0.001410
0.001417
```

tatulea, 2019-07-09:
I am not too much into math and this is why I don't know exactly how to solve this. My final point should be the fact that the `x` input that I am using should return me a peak frequency of about 200MHz

dmitry1945, 2019-07-09:
dsps_tone_gen_f32(x, N, 1.0, 0.16,  0); is an error. You overwrite your input signal with tone.
Remove this line.

tatulea, 2019-07-10:
I tried, but I still got the wrong values

this-username-is-taken, 2020-01-03:
> Hi Gabriel,
> The esp-dsp library working with complex numbers, that's why the simplest way for you will be just ignore the x2 input and y2 output, and use only x1 and y1. Your code will look so: [...]

This is great! This should be in the documentary. Made things so much easier to understand.

Btw, why is there not a real-input-only implementation of FFT?

nakanaela, 2020-01-17:
From my experience with rFFT algorithms in the CMSIS DSP library, there really is no such thing as a 'real' only fft. Those functions do a bit of hand-waving to allow you to use a vector of real only inputs and creates a secondary vector of 2*sample size with every i*2+1 index set to zero to account for the complex portion. The benefit is that you don't have to do it yourself, but the resulting firmware files were easily twice what I get from zeroing the complex index myself and directly using the corresponding cFFT function. 

If you are sampling real data, it is pretty easy to account for the complex value being '0' and in my experience has been faster than refactoring a vector of real inputs to a vector of real+cmplx values. If you initialize your input array to '0' from the start, then all you have to worry about is adding your real inputs at every [2*sample#] as [2*sample#+1] should already be '0'.

drewandre, 2020-03-15:
Hi @dmitry1945 - I was hoping you could elaborate on the fft example a bit more in the context of audio streaming. I have the example referred to above working great but I'd like to use the audio stream from an esp-adf i2s stream and I don't see any examples in any esp library nor have I seen any helpful forum posts which is surprising.

At a high level, I'm trying to stream audio to the esp32 lyrat 4.3 board, run an fft, and display the spectrum across an led strip. I have each element working in isolation (the bluetooth audio stream, writing i2s data to the lyrat audio codec, outputting audio from the codec, running non-audio reactive led animations, and running the above fft example).  

I've tried a few approaches including 1) trying to read the i2s raw byte array and iterating over that, 2) creating a audio pipeline and passing the byte array over to the fft function, and 3) implementing an i2s write callback.

Each of these either doesn't compile or work, but I'm also not sure if there's something obvious I'm missing or if I'm not on the right track at all. Any pointers here on what methods I could leverage from the esp-adf library to pass the i2s buffer over to the `dsps_fft2r_fc32` function?

Thanks!

dmitry1945, 2020-03-16:
Hi @drewandre , 

If I understood correct, you have next application:
I2S_read/bluetooth read, fft, some adf codec, I2S_write. 
I would suggest to try next: prepare the buffer with `dsps_tone_gen_f32` and pass this buffer as an predefined input to the `dsps_fft2r_fc32` inside your application instead of buffer with real data. In this case you should see the expected result. If  you see the expected result, then we know, that the problem is not in the application and in functions calls, but in the data format. If the problem in data format, then just convert your input to the float and use  or use `dsps_fft2r_fc32` as in example, or use `dsps_fft2r_sc16` instead.
If you don't see the expected result with predefined data, then the problem in the application and you have to care about execution flow. Please let me know if it so, I will provide suggestions. 

Regards,
Dmitry

drewandre, 2020-03-17:
Thanks for the quick response @dmitry1945. You are correct that the application is:
```
                          |-> FFT from i2s stream -> LED animations from FFT data
bluetooth audio stream -> |
                          |-> i2s write -> audio codec headphone output
```
I have already been using `dsps_tone_gen_f32` as you suggested and it works well. I was using this just to make sure the fft logic is running correctly.

This question is actually more of an esp-adf library question so I opened [an issue](https://github.com/espressif/esp-adf/issues/365) on the esp-adf page. But I'm wondering if you have any pointers on how to pass an i2s stream to `dsps_fft2r_fc32`. I assumed this library would play a little nicer with esp-adf like the teensy audio library which has an i2s audio element that can be ["patched" directly to the fft](https://github.com/PaulStoffregen/Audio/blob/master/examples/Analysis/FFT/FFT.ino#L32) but it's clear that these are two completely separate libraries. For example, the fft functions in esp-dsp accept floating or fixed buffers, but the audio element in esp-adf [works with char buffers](https://docs.espressif.com/projects/esp-adf/en/latest/api-reference/framework/audio_element.html#_CPPv412process_func).

The repo I'm working on is basically the [a2dp_gatts_coex esp-idf example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex). I won't be using the fft for audio processing - I would be using it to control LED animations so the fft wouldn't have to be part of the audio pipeline.

So how would you suggest I use this library with esp-adf's audio element char buffer stream? If you feel this question is better for esp-idf, feel free to close it this. I would appreciate any tips!

dmitry1945, 2020-03-19:
Hi @drewandre ,

I think the best way to do FFT is to run it on separate thread. 
Make thread with semaphore, and when data comes to the I2s stream callback just copy the i2s buffer to the fft processing buffer, and then trigger the semaphore if the FFT thread is not busy.
Something like this:
```
bool fft_busy = false;
byte data_for_fft[NNN];
void some_i2s_callback()
{
     ... fill buffer with data for fft - data_for_fft
     if (false == fft_busy) xSemaphoreGive(fft_task_sem);
     .... some other processing
}
void FFT_thread()
    while (1)
    {
        fft_busy  = false;
        xSemaphoreTake(fft_task_sem, 10000000);
        fft_busy = true;
        .... here convert i2s buffer (data_for_fft) to fft complex float
        .... process buffer in fft and calculate magnitude
        .... show buffer
    }
}
```

Regards,
Dmitry

drewandre, 2020-03-19:
Thanks @dmitry1945, I think I'm pretty close (haven't implemented threading yet) but `dsps_view` is showing some weird data. You can view my dsps_view output [here](https://gist.github.com/drewandre/189637a3705bc41021981aef11ce02e3). Do you mind taking a look at the code snippets below and see if anything looks incorrect? Am I not properly casting the uint8_t buffer array to float buffer?

For running an esp-dsp FFT on an a2dp bluetooth audio stream:

1. Initialize fft
```
#define BUF_SIZE 4096
void init_fft()
{
  esp_err_t ret = dsps_fft2r_init_fc32(NULL, BUF_SIZE);
  if (ret != ESP_OK) { ... }
}
```
2. Initialize A2DP sink with default `bt_app_a2d_cb` config and `bt_app_a2d_data_cb` data callback
```
esp_a2d_register_callback(&bt_app_a2d_cb);
esp_a2d_sink_register_data_callback(bt_app_a2d_data_cb);
esp_a2d_sink_init();
```
3. Connect an fft data handler function in `bt_app_a2d_data_cb`
```
void bt_app_a2d_data_cb(const uint8_t *data, uint32_t len)
{
  size_t bytes_written;
  i2s_write(0, data, len, &bytes_written, portMAX_DELAY);
  copy_a2dp_buffer_for_fft(&data, len); // len = 4096
}
```
4. Copy data over to a new buffer in `copy_a2dp_buffer_for_fft`
```
static uint8_t fft_buffer[BUF_SIZE];
void copy_a2dp_buffer_for_fft(const uint8_t *data, uint32_t len)
{
  memcpy(fft_buffer, data, len);
}
```
5. Run `calculate_fft` function which can be viewed in this [gist](https://gist.github.com/drewandre/3f4ecedc07b85fd77a1b3f156e9e43ff)

Thanks again for your help. Hopefully with this running I could post a full working example of using the esp-dsp fft with audio streams.

drewandre, 2020-03-30:
Hi @dmitry1945, thanks again for your help. I think I'm pretty close to getting esp-dsp's FFT's running against a bluetooth audio stream (interleaved PCM stereo audio) but `dsps_view`'s serial output just doesn't look right even with a pure 440hz sine wave. I'm still running ESP-IDF 3.3.1 and ESP-ADF v2.0-beta3.

The serial output of `dsps_view` can be viewed [here](https://gist.github.com/drewandre/fa1e024d5a193ff963bdbf19ea0b9d6d), and the float buffer passed to `dsps_fft2r_fc32` and related functions can be viewed [here](https://gist.github.com/drewandre/a87e45d6a9a38608e2188c64b676fcf6).

For reference, [here](https://gist.github.com/drewandre/87a20dd40f4f0de4a6f92a6e35a7e31a) is `dsps_view`'s serial output when swapping out my PCM buffer with the one generated by `dsps_tone_gen_f32(x1, N, 1.0, 0.16, 0)` (only copying this here to show that the FFT logic is correct - it just seems to be a data format issue with my PCM buffer). I do notice here that the the `min` property printed by `dsps_view` is -91.273849 for my PCM 440hz sine tone, and -inf when using `dsps_tone_gen_f32`'s output. Unsure if this is relevant.

Do you mind checking my fft logic to see if anything jumps out? fft_controller.c in full can be viewed on my github repo [here](https://github.com/drewandre/palette_firmware_esp32/blob/feature/arduino-fft/main/fft_controller.c). At a high level, here is what I'm doing:

1. Initialize an a2dp data callback to copy the outgoing buffer over to `fft_buffer` to be used for FFT analysis:
```
#define N_SAMPLES 2048
uint8_t fft_buffer[N_SAMPLES * 2] // 4096 because that's the length of the esp-adf's a2dp callback buffer

// data is a pointer to a uint8_t array of PCM interleaved stereo audio data
// `len` is always 4096 here
void copy_a2dp_buffer_to_fft_buffer(const uint8_t *data, uint32_t len)
{
  if (!fft_running)
  {
    memcpy(fft_buffer, data, len);
  }
}
```
2. Initialize fft, dynamically allocate some buffers, and fill `wind` and `x1` with window and tone data (`dsps_tone_gen_f32` doesn't really need to be called, just added it to init a buffer of data that we know is correct). **I assume dynamically allocating these arrays as well as passing `NULL` as a table_buff parameter is ok? I also assume use of calloc over malloc wouldn't be an issue.**
```
float *x1, *wind, *y_cf, *y1_cf;

void init_fft()
{
  esp_err_t ret = dsps_fft2r_init_fc32(NULL, CONFIG_DSP_MAX_FFT_SIZE);
  x1 = (float *)calloc(N_SAMPLES, sizeof(float));
  wind = (float *)calloc(N_SAMPLES, sizeof(float));
  y_cf = (float *)calloc(N_SAMPLES * 2, sizeof(float));
  y1_cf = &y_cf[0];

  dsps_wind_hann_f32(wind, N_SAMPLES);
  dsps_tone_gen_f32(x1, N_SAMPLES, 1.0, 0.16, 0);
}
```
3. Attach the fft task (I haven't been using semaphores yet):
```
xTaskCreatePinnedToCore(&calculate_fft, "calculate_fft", 4000, NULL, 5, NULL, 1);
```
4. Establish an a2dp connection and begin streaming audio
5. In `calculate_fft` which runs every 15 seconds, fill `x1` float array with formatted values from `fft_buffer`.  The simple `map` function used can be viewed [here](https://gist.github.com/drewandre/0bfcb76e5ad785308ea5ec740d857bf8). **This is likely where the issue is since this is where the data is actually mapped to the format `dsps_fft2r_fc32` expects:**
```
fft_running = true; // set flag which stops the memcpy in step 1

for (int i = 0; i < N_SAMPLES; i++) {
  // i * 2 since stereo interleaved buffer is structured [left][right][left][right]...
  // so in this example we are only building an fft buffer for the left channel.
  // we then divide the sample by 255.0 to reduce the range to 0 - 1.
  float sample = (float)fft_buffer[i * 2] / 255.0;

  // here we map the sample from 0 - 1 to -1 to 1.
  x1[i] = map(sample, 0, 1, -1, 1);
}
```
5. Run the fft (same exact logic as the [fft example](https://github.com/espressif/esp-dsp/tree/master/examples/fft) but with a larger `dsps_view` output):
```
// Convert two input vectors to one complex vector
for (int i = 0; i < N_SAMPLES; i++)
{
  y_cf[i * 2 + 0] = x1[i] * wind[i];
  y_cf[i * 2 + 1] = 0;
}

dsps_fft2r_fc32(y_cf, N_SAMPLES);
// Bit reverse
dsps_bit_rev_fc32(y_cf, N_SAMPLES);
// Convert one complex vector to two complex vectors
dsps_cplx2reC_fc32(y_cf, N_SAMPLES);

for (int i = 0; i < N_SAMPLES / 2; i++)
{
  y1_cf[i] = 10 * log10f((y1_cf[i * 2 + 0] * y1_cf[i * 2 + 0] + y1_cf[i * 2 + 1] * y1_cf[i * 2 + 1]) / N_SAMPLES);
}

dsps_view(y1_cf, N_SAMPLES / 2, 128, 20, -60, 40, '|');

fft_running = false; // set flag to allow memcpy to copy PCM buffer to fft_buffer in step 1

vTaskDelay(15000 / portTICK_PERIOD_MS); // only run every 15 sec for debugging
```

Anything look incorrect to you? I understand this may be more of an issue with how I'm handling the PCM buffer and not an issue with the library, so feel free to just say you're not sure and I'll go ahead and post on some forums. That being said, it would be great to figure this out so other users could more easily get esp-adf and esp-dsp to play nicely together :) Thanks in advance!

dmitry1945, 2020-03-31:
Hi @drewandre, 
Just to be sure, how you are initialize the y1_cf value? 
I think in our case it's possible to use next:
`y1_cf[i] = 10 * log10f((y_cf[i * 2 + 0] * y_cf[i * 2 + 0] + y_cf[i * 2 + 1] * y_cf[i * 2 + 1]) / N_SAMPLES);
`

drewandre, 2020-03-31:
@dmitry1945 Thanks for the tip - I've made this adjustment.

I have a few quick questions for you if you don't mind -
1. How come you suggested using semaphores over just a regular FreeRTOS task?
2. Rather than using `dsps_bit_rev_fc32`, can I just reverse my for loop in step 5?
3. And regarding memory management, would you suggest passing my x1 buffer to `dsps_fft2r_init_fc32` rather than having it internally allocated?

dmitry1945, 2020-04-02:
Hi @drewandre ,
I will try to answer,

1. The semaphore hold the FFT task until event will come. It just synchronize your BT callback and FFT calculation.
2. The dsps_bit_rev_fc32 is not just a loop. It's doing some more complicated work. For example if length is 256, bit reverse of 10000000b is 00000001b, 00000010b is 01000000b, 00000110b is 01100000b, and so on.
3. I think better to have as it is. 

Regards,
Dmitry

wired8, 2022-07-08:
Hi @drewandre did you ever get this working?   I'm looking to calculate 1/3 octaves from a streaming source.

drewandre, 2022-07-10:
Yes, it has been over a year since I've worked on the project but I've attached code samples that may be helpful.

audio_analysis_controller.cpp: https://gist.github.com/dandre-hound/952127e65cb5565e3928c5121977132d

audio_analysis_controller.hpp: https://gist.github.com/dandre-hound/e4357b422a144fb8f4ba69ee0a8bdf2a

There’s a lot of extra code in here unrelated to what you’re trying to do, but to calculate octave bands, I created an initial octave_bands array, and then used the pre-defined frequency bounds to group the outputs of the fft analysis. Relevant functions in these files include  average_into_octave_bands, calcAvg, and freqToIndex.

wired8, 2022-07-16:
Thanks for that @drewandre much appreciated.

closed 2023-07-28 by dmitry1945