espressif/esp-dsp #6: Enhancement: ESP-DSP *really* needs high-performance sin()/cos()

danak6jq, 2019-09-20:
I've implemented a DSP application in an ESP32 using ESP-DSP (using vector math and FIRs), and the performance is _very_ good, comparable to STM32F446x using CMSIS-DSP when scaled for clock frequency - very impressed! 

However, one thing that DSP applications often do is mix a signal with an I/Q 'oscillator' implemented as in dsps_tone_gen_f32(), except both the sin() and cos() are calculated together. I noticed the performance of the sin() and cos() library functions is not optimized and very poor. My estimate is that sin() and cos() each take ~2060 clocks per call, a total of ~4120 clocks per I/Q calculation.

As an experiment, I built the ARM CMSIS-DSP optimized arm_sin_f32() and arm_cos_f32() and these take ~225 clocks per call, a total of ~450 clocks per I/Q calculation, a little greater than 9x the performance.

ESP-DSP would be *greatly* enhanced with the addition of optimized sin()/cos() functions, perhaps even as a single function that returns both the sin() and cos() of the same input.

Thank you!
Dana

dmitry1945, 2019-09-20:
Hi Dana,
Thank you for feedback.

The dsps_tone_gen_f32() was not designed to be optimized. is made to generate table in non real-time thread, or in initialization phase.

Anyway, I think we will implement optimized sin/cos functions soon.

Thank you for the feedback!
Dmitry

danak6jq, 2019-09-20:
Thank you, Dmitry - this is terrific news. I expected dsps_tone_gen_f32() was not intended as an optimized function, I was just using it as an example of an NCO. Optimized sin()/cos() would be terrific.

pipehappy1, 2021-07-15:
I wonder any updates on this issue? fast sin/cos would be great!

danak6jq, 2021-07-19:
In my application, I evaluated the license terms of the CMSIS-DSP source and used the fast cos/sin function from that until ESP-DSP has this functionality

dmitry1945, 2023-07-28:
Hi @danak6jq the optimyzed sin/cos generator now in the esp-dsp.

closed 2023-07-28 by dmitry1945