espressif/esp-dsp #11: Possible to use ESP-DSP in Arduino for ESP32?

ohnoitsalobo, 2020-02-01:
My questions:
1) is it possible to use `ESP-DSP` in Arduino code, similar to other `ESP-IDF` code?
2) if yes, can you make a basic example I could follow? 

Hello, I have made a working spectrum analyzer using `ArduinoFFT` [(link)](https://github.com/kosme/arduinoFFT) which is obviously not optimized for ESP32. Similar to [this man's work](https://github.com/G6EJD/ESP32-8266-Audio-Spectrum-Display/). In my case I use it to drive an LED strip.

I generally do not use `ESP-IDF` as it is very confusing for me, however I am aware that some `ESP-IDF` functions are still accessible in the `arduinoespressif32` framework. Is this possible with `ESP-DSP`?

I am actually attempting to use it right now, but as I am not a professional programmer my progress is very slow. A lot of compiler errors and definition errors etc. It will probably take me a week to do (badly) what you can do (properly) in an hour. So my request is: can you make an example using `ESP-DSP` FFT in Arduino? Similar to [this `ArduinoFFT` example](https://github.com/kosme/arduinoFFT/blob/master/Examples/FFT_03/FFT_03.ino), reading values from an analog pin and running FFT on them.

rtufino, 2020-05-10:
Hi. Can you solve this?

ohnoitsalobo, 2020-07-07:
I wasn't able to get it to work.
Hoping someone smarter than me weighs in eventually.

hamuro80, 2020-09-23:
I can use esp dsp under esp-isd with arduino as a component, but can't use it under arduino ide. I think someone should port the esp-dsp component to arduino library, any clue how to start?

standarddeviant, 2020-10-29:
Is a PlatformIO project configured to use the Arduino framework a suitable solution? Their build flags system is flexible which helps. I think I've done this before and I'll post an example if I can confirm that I have it working.

standarddeviant, 2020-10-29:
It's clunky, but have a look at this example that uses the biquad assembly function:
https://github.com/standarddeviant/Demo_ESP32_PlatformIO_Arduino_ESPDSP/tree/main/lib

That repo builds for me. I moved source files in to `lib/esp-dsp-pio/src` and header files to `lib/esp-dsp-pio/include`.

Some of the FFT source/header files have issues with how they define some const arrays. You should be able to use ESP-DSP's FFT functions, but it will probably require some minor modifications of the FFT functions. I'll add to that repo and try to add the FFT functions.

michikite, 2020-11-11:
you can use platformIO and build with both esp-idf and Arduino. The component is placed in a 'components' folder.
Just tried that with dotprod example and it works fine.

Example settings:

platformio.ini
```
[env]
platform = espressif32
framework = 
	arduino
	espidf
monitor_speed = 115200
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0
```
CMakeLists.txt
```
cmake_minimum_required(VERSION 3.16.0)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
list(APPEND EXTRA_COMPONENT_DIRS esp-dsp)
project(espidf-arduino-example)
```

me-no-dev, 2020-11-23:
it's coming in Arduino 2.0.0

kylongmu, 2021-04-03:
> 
> 
> it's coming in Arduino 2.0.0

Where can see the message with Arduino 2.0 ?

closed 2022-03-07 by igrr