espressif/esp-dsp #13: Fixed-point FFT is slower than floating point FFT

wzli, 2020-03-01:
Hi,
Is there any official benchmarks for the speed of fixed point FFTs?https://docs.espressif.com/projects/esp-dsp/en/latest/esp-dsp-benchmarks.html does not seem to be updated.

I've been testing myself on a AI-Thinker ESP32-CAM board, and my results for N=64 show that the sc16 FFT is slower than fc32 FFT by about 35%. Is that expected performance or did I do something wrong?

dmitry1945, 2020-03-02:
Hi wzli,

Yes, it's expected performance.
And the reason for that is next:
In Esp32 we have floating point unit. And for floating point operation and for fixed point operation we will spend the same amount of add and multiply operations. But, for fixed point, to have acceptable SFDR, you have to make additional job like rounding. 
You can look to the implementation of fixed point FFT in ansi C. You will see the difference.

Regards,
Dmitry

wzli, 2020-03-02:
I see. I was under the impression that there would be SIMD instructions optimized for short integers (like the ARM Cortex-M DSP extentions). Otherwise there seems to be no reason to support 16-bit fixed point in the first place, compared to the float implementation which is better in general. In any-case, please posts the benchmarks for future reference. Thanks.

closed 2020-03-02 by wzli