espressif/esp-dsp #14: esp-dsp library as component of esp-idf project

marina936, 2020-06-24:
I want to use these libraries in my esp-idf projects. I am using PlatformIO framework of Visual Code to write my code.
As I see in the PlatformIo website https://docs.platformio.org/en/latest/frameworks/espidf.html#esp-idf-components, I clone dsp libraries in my project as a component and I add is CmakeList.txt the path to the component for including the libraries:
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
list(APPEND EXTRA_COMPONENT_DIRS esp-dsp)
project(subscribe_publish)

When I  build the project I see this error 

-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
CMake Error at C:/Users/.../.platformio/packages/framework-espidf/tools/cmake/project.cmake:277 (set):
  Maximum recursion depth of 1000 exceeded
Call Stack (most recent call first):
  C:/Users/.../.platformio/packages/framework-espidf/tools/cmake/project.cmake:278 (__project)

This line  C:/Users/.../.platformio/packages/framework-espidf/tools/cmake/project.cmake:278 (__project) refers to this  line of thee file project.cmake:
function(project)
        set(project_ARGV ARGV)
        __project(${${project_ARGV}})

dmitry1945, 2020-06-24:
Hi @marina936,
You have to do this in different way.

For example you have your project: my_example.
You have some folder with esp-dsp and other extra components for esp-idf. This is your extra components directory.
Inside your project you have CMakeLists.txt
Basically this file contain something like this:

```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(my_example)
```
You have to add to this file a lines which will define path to your extra components directory

`set(EXTRA_COMPONENT_DIRS "PATH to your extra components directory")`

Finally your CMakeLists.txt will looks like this:

```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS "PATH to your extra components directory")

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(my_example)
```
I think it will work. 

Regards,

Dmitry

marina936, 2020-06-24:
If I do that I have this error: Error: Couldn't find the main target of the project!

dmitry1945, 2020-06-24:
@marina936 is the project structure like for example in the esp-dsp/examples/fft ? With main folder and with main() source and other files (sure it's just example)? 
What is your current project structure? Is it looks similar to the "esp-dsp/examples/fft"?

marina936, 2020-06-24:
Yes I have this project structure 
[image]

dmitry1945, 2020-06-24:
OK. 
In your src folder you should have CMakeLists.txt file like this:
```
set(COMPONENT_SRCS "my_project_source_with_main.c")
set(COMPONENT_ADD_INCLUDEDIRS "../include")

register_component()
```
In this file you have to list all source files that you want to use from /src folder. 

In your root folder you should have CMakeLists.txt file that I have describe before.

marina936, 2020-06-24:
In CMakeLists.txt of src, I have this: 

FILE(GLOB_RECURSE app_sources ${CMAKE_SOURCE_DIR}/src/*.*)
idf_component_register(SRCS ${app_sources})

I try to add this line: 
set(COMPONENT_ADD_INCLUDEDIRS "../components")

Like:
FILE(GLOB_RECURSE app_sources ${CMAKE_SOURCE_DIR}/src/*.*)
set(COMPONENT_ADD_INCLUDEDIRS "../components")
idf_component_register(SRCS ${app_sources})

but I have the same error

dmitry1945, 2020-06-24:
@marina936 I think you trying to mix up standard cmake rules and cmake rules for the esp-idf. It's not always one-to-one.
Can you please try to copy cmake files structure from examples (for example fft), where all source files must be listed or with one simple main.c?
I would recommend to place in root folder CMakeLists.txt file just with this content:
```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

# This line is optional
set(requires esp-dsp)
# This line is required
set(EXTRA_COMPONENT_DIRS "./components")

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(my_example)
```
and to your src folder file with this content:

```
set(COMPONENT_SRCS "file_with_main.c"
"some_other_file1.c"
"some_other_file2.c"
...
"last_file_of_your_project.c"
)

set(COMPONENT_ADD_INCLUDEDIRS "../include")

register_component()
```

closed 2020-07-17 by marina936