espressif/esp-dsp #16: How do I run the test suite?

brianbienvenu, 2020-09-23:
I tried to compile and run the "test_app" included here, however it just says:
```
Here's the test menu, pick your combo:

Enter test for running.
```
I enter a number, and get:
```
-----------------------
0 Tests 0 Failures 0 Ignored
OK
Enter next test, or 'enter' to see menu
```

I guess that the files in the test_app folder don't like up with the test suite correctly, how do I make it work?

dmitry1945, 2020-09-23:
I think now you have place esp-dsp to the `esp-edf/components/esp-dsp` folder. But for tests, you have to place esp-dsp to the `esp-edf/components/dsp` folder. Just rename it.
Then in two files you will need to removw lines `#include "fft.h"` (this is bug. A bugfix will come soon).

Regards,
Dmitry

dmitry1945, 2020-09-23:
@brianbienvenu just rename filder. 
The headers I've removed.

Regards,
Dmitry

brianbienvenu, 2020-09-24:
Thanks, it works after changing the folder name, and with your new commit.

Should "how to run the tests" instructions be added to the repo readme, or should this have been obvious to me?

dmitry1945, 2020-09-24:
"How to run the tests" is the standard procedure for the esp-idf and it's described there. Or you mean something else?
I will add to read.me description with renaming.

Regards,
Dmitry

brianbienvenu, 2020-09-24:
Thanks, no worries. I'm not an esp-idf expert yet!

closed 2020-09-24 by brianbienvenu