espressif/esp-dsp #18: ESP32-S2 support?

nikolindima, 2020-10-08:
Hello. Will you add ESP32-S2 support for ESP-DSP?

dmitry1945, 2020-10-12:
Hi @nikolindima ,
Current version of ESP-DSP support Esp32-S2 chip.
The Esp32-S2 has no FPU inside, and has no additional multipliers, that's why ESP-DSP contain for Esp32-S2 only ANSI realizations.

Regards,
Dmitry

askpatrickw, 2020-10-26:
Suggestion: Somewhere in the Readme explain this DSP library (What does DSP stand for?) and its relationship to the ESP32-S2 DSP -- Digital Signature Peripheral

dmitry1945, 2020-10-26:
Hi @askpatrickw ,
The DSP is a common abbreviation for Digital Signal Processing. 
I will look to readme and explain it.

Thank you.
Regards,
Dmitry

me-no-dev, 2020-11-23:
@dmitry1945 it seems that even if ANSI is selected, all sources are compiled, so it fails on ESP32-S2
```
 -c ../components/esp-dsp/modules/dotprod/float/dsps_dotprod_f32_ae32.S
../components/esp-dsp/modules/dotprod/float/dsps_dotprod_f32_ae32.S: Assembler messages:
../components/esp-dsp/modules/dotprod/float/dsps_dotprod_f32_ae32.S:45: Error: unknown opcode or format name 'wfr'
../components/esp-dsp/modules/dotprod/float/dsps_dotprod_f32_ae32.S:50: Error: unknown opcode or format name 'lsx'
../components/esp-dsp/modules/dotprod/float/dsps_dotprod_f32_ae32.S:50: Error: unknown opcode or format name 'loopnez'
../components/esp-dsp/modules/dotprod/float/dsps_dotprod_f32_ae32.S:50: Error: unknown opcode or format name 'lsx'
../components/esp-dsp/modules/dotprod/float/dsps_dotprod_f32_ae32.S:50: Error: unknown opcode or format name 'madd.s'
../components/esp-dsp/modules/dotprod/float/dsps_dotprod_f32_ae32.S:50: Error: unknown opcode or format name 'lsx'
../components/esp-dsp/modules/dotprod/float/dsps_dotprod_f32_ae32.S:52: Error: unknown opcode or format name 'ssi'
```

sdkconfig:
```
#
# DSP Library
#
CONFIG_DSP_ANSI=y
# CONFIG_DSP_OPTIMIZED is not set
CONFIG_DSP_OPTIMIZATION=0
# CONFIG_DSP_MAX_FFT_SIZE_512 is not set
# CONFIG_DSP_MAX_FFT_SIZE_1024 is not set
# CONFIG_DSP_MAX_FFT_SIZE_2048 is not set
CONFIG_DSP_MAX_FFT_SIZE_4096=y
# CONFIG_DSP_MAX_FFT_SIZE_8192 is not set
# CONFIG_DSP_MAX_FFT_SIZE_16384 is not set
# CONFIG_DSP_MAX_FFT_SIZE_32768 is not set
CONFIG_DSP_MAX_FFT_SIZE=4096
# end of DSP Library

CONFIG_C_IMPL=y
# CONFIG_XTENSA_IMPL is not set
```

dmitry1945, 2020-11-23:
@me-no-dev 
I will provide update this-next week.
Thanks for remind.

Regards,
Dmitry

deets, 2020-12-03:
Bump. I also ran into this.

dmitry1945, 2020-12-03:
Hi @deets 
It's in progress...
Will release soon.

Regards,
Dmitry

dmitry1945, 2020-12-14:
@nikolindima @me-no-dev @deets 
Now master contain support for Esp32-s2

Regards,
Dmitry

nikolindima, 2020-12-14:
@dmitry1945 this is only ANSI DSP?

dmitry1945, 2020-12-14:
@nikolindima  Esp32-S2 has no FPU and has no some other units inside. That's why will be no big difference between asm and ANIS.

Regards,
Dmitry

nikolindima, 2020-12-14:
@dmitry1945 Thanks

douardda, 2021-01-28:
It looks there is //some// support for the esp32s2, but it's unclear what is actually supported and what is not. For example, I have a few  `undefined reference to `dsps_biquad_f32_ae32'` when I try to compile (well link at this point) a project for the esp32s2.

A bit frustrating.

dmitry1945, 2021-01-29:
Hi @douardda 
The dsps_biquad_f32_ae32 should not be called directly from the source. It used under wrapper name dsps_biquad_f32. And this name is handled by esp-dsp depends on Esp32/Esp32-S2. For Esp32 it's possible to use optimized version which is dsps_biquad_f32_ae32, and for S2 it's possible to use only dsps_biquad_f32_ansi.

By the way, in which project or file do you have "undefined reference to dsps_biquad_f32_ae32"?

Regards,
Dmitry

douardda, 2021-01-29:
Hi @dmitry1945

I'm trying to compile @jorgenkraghjakobsen project  https://github.com/jorgenkraghjakobsen/snapclient for my ESP32-S2-Saola board (using a pcm5142 DAC).

I've replaced the esp-dsp that comes with it (which had no support for s2 back then) with the current version from github, and it compiles ok, but linking fails with this undefined reference, because the embedded component esp_processor use this function directly. Thanks for the tip on using `dsps_biquad_f32` instead, I'll give a try.

(note that meanwhile I've removed all code related to signal processing in the project, thus the need for biquad filtering) and could finally compile and link it, and I have other problems at execution time while attempting to create a ringbuffer in psram, but meh).

jorgenkraghjakobsen, 2021-01-29:
Hi Just ordered an ESP32-S2 based board and will look into the issue as well. My snapcast project is very much WIP. I faced  problem related to usage of dsps_biquad.... from non ADF env and imported want I needed and coded on. I will try to find a more elegant way to solve this issue. /J

douardda, 2021-01-29:
@jorgenkraghjakobsen I'm trying to write an adp "version" of your snapclient (using my esp32s2), but I face difficulties using adp for this. I can't get opus nor ogg decoders provided in esp-adf (but unfortunately [not open-source](https://github.com/espressif/esp-adf-libs/issues/12) ). I have flac and pcm "working" (with lots of glitches, not sure why, maybe the required bw is too high, and no latency control at all).

I've put the "code" on https://github.com/douardda/snapclient-esp-adf if you want to have a look.

jorgenkraghjakobsen, 2021-01-29:
I do not use the opus/ogg coder from the ADF - I use the open soured version with a ESP32 wapper around - I will be very happy to have a look on your flac implementation as well. There are som settings you can do on you wireless router / esp wifi setup to disable package aggregation - to optimize latency vs through put. 
On my site is had a little brake through on the timing control - think I will be ready to upload next week. 
/J

igrr, 2022-03-07:
Since ESP32-S2 support has been available for some time, I'm going to close this issue.

closed 2022-03-07 by igrr