espressif/esp-dsp #19: dsps_cplx2real_fc32 cannot be used without fft4r initialization

KaSroka, 2020-11-23:
In my application I'm using radix-2 FFT (I'm using 4096 real points so FFT is 2048 points).
Current implementation of dsps_cplx2real_fc32 requires fft radix-4 to be initialized to work properly.

Why there is no implementation for radix-2 alone? Having to init radix-4 wastes 32kB of RAM in my application for tables that are only needed for dsps_cplx2real_fc32.

dpavlovich, 2020-11-23:
Hi @KaSroka , 
it should be possible. I will check. Will update if found something.

Regards,
Dmitry

dmitry1945, 2020-11-23:
Hi @KaSroka ,
it should be possible. I will check. Will update if found something.

Regards,
Dmitry

KaSroka, 2020-11-23:
`dsps_cplx2real_fc32` is defined as `dsps_cplx2real_fc32_ae32` when using `CONFIG_DSP_OPTIMIZED`. `dsps_cplx2real_fc32_ae32` is defined as a call to `dsps_cplx2real_fc32_ansi_` with `dsps_fft4r_w_table_fc32` as `table` and `dsps_fft4r_w_table_size` as `table_size`. These is no version that uses radix-2 tables. When explicitly calling `dsps_cplx2real_fc32_ae32_` with `dsps_fft_w_table_fc32` as `table` and `dsps_fft_w_table_size` as `table_size` the algorithm doesn't work as it expects radix-4 table which is bigger. There is no definition of `dsps_cplx2real_fc32` in `dsps_fft2r_fc32*` files, there is only `dsps_cplx2reC_fc32` but it's for other scenarios. Also ANSI version of `dsps_cplx2real_fc32` checks if radix-4 FFT is initialized.

KaSroka, 2020-11-23:
Also the docstring for `dsps_cplx2real_fc32_ae32_` is wrong as it mentions two inputs and this is responsibility of `dsps_cplx2reC_fc32`

dmitry1945, 2020-12-04:
Hi @KaSroka 

Yes, the dsps_cplx2real_fc32(...) can be used only if FFT4R was initialized because this function use FFT4R tables. I  have add the description about this, and release will be soon.
If you use this function you have to initialize FFT4R, because the allocated size of memory will be almost the same, and FFT4R is ~30 % faster then FFT2R. If the time is very critical, better to use FFT4R.
To have FFT2R with real input it's better to use dsps_cplx2reC_fc32 and two real inputs at the same time.

Thanks,
Dmitry

closed 2023-07-28 by dmitry1945