espressif/esp-dsp #20: Current master branch example code is not compiling

ashvarma-git, 2020-12-15:
I did a git clone and make of dotprod example as per README.md

    git clone https://github.com/espressif/esp-dsp.git
    cd esp-dsp/examples/dotprod
    make

I get lots of compile errors of this type:

.../GitHub/esp-dsp/modules/fft/float/dsps_fft2r_bitrev_tables_fc32.c:579:5: error: initializer element is not constant
     (uint16_t)bitrev2r_table_16_fc32_size,

I use ESP-IDF 3.3.1 and toolchain:

% xtensa-esp32-elf-gcc -v
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/ashvarma/GitLab/CTR-01-Rel2/src/toolchain/xtensa-esp32-elf-80/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-sysroot=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-80-g6c4433a' --disable-__cxa_atexit --enable-cxx-flags='-fno-rtti -ffunction-sections' --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-cloog=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-libelf=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-libgomp --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-nls --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio
Thread model: posix
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a)

dmitry1945, 2020-12-15:
Hi @ashvarma-git 
Could you please try change (uint16_t) to (const uint16_t) in table initialization. I think it will help.
In this case I will commit changes to master.
```

const uint16_t dsps_fft2r_rev_tables_fc32_size[] = {
    (const uint16_t)bitrev2r_table_16_fc32_size,
    (const uint16_t)bitrev2r_table_32_fc32_size,
    (const uint16_t)bitrev2r_table_64_fc32_size,
    (const uint16_t)bitrev2r_table_128_fc32_size,
    (const uint16_t)bitrev2r_table_256_fc32_size,
    (const uint16_t)bitrev2r_table_512_fc32_size,
    (const uint16_t)bitrev2r_table_1024_fc32_size,
    (const uint16_t)bitrev2r_table_2048_fc32_size,
    (const uint16_t)bitrev2r_table_4096_fc32_size,
};

```

Thanks,
Dmitry

ashvarma-git, 2020-12-15:
Dmitry,

Still get errors with this suggested change -

esp-dsp/modules/fft/float/dsps_fft2r_bitrev_tables_fc32.c:579:5: error: initializer element is not constant
     (const uint16_t)bitrev2r_table_16_fc32_size,

esp-dsp/modules/fft/float/dsps_fft2r_bitrev_tables_fc32.c:579:5: note: (near initialization for 'dsps_fft2r_rev_tables_fc32_size[0]')

The problem is that as per C language standard, a const data type can only have a literal constant value. A const cannot be initialized by another const data type.

So if I replace this line with the literal value:

    (const uint16_t) 6,

It compiled OK for this line but then fails for next one in the dsps_fft2r_rev_tables_fc32_size[] array.

Please check this discussion about C compile standard:

https://stackoverflow.com/questions/3025050/error-initializer-element-is-not-constant-when-trying-to-initialize-variable-w

dmitry1945, 2020-12-15:
@ashvarma-git 
I've replace:
```
const uint16_t dsps_fft2r_rev_tables_fc32_size[] = {
    (const uint16_t)6,   // bitrev2r_table_16_fc32_size,
    (const uint16_t)12,  // bitrev2r_table_32_fc32_size,
    (const uint16_t)28,  // bitrev2r_table_64_fc32_size,
    (const uint16_t)56,  // bitrev2r_table_128_fc32_size,
    (const uint16_t)120, // bitrev2r_table_256_fc32_size,
    (const uint16_t)240, // bitrev2r_table_512_fc32_size,
    (const uint16_t)496, // bitrev2r_table_1024_fc32_size,
    (const uint16_t)992, // bitrev2r_table_2048_fc32_size,
    (const uint16_t)2016,// bitrev2r_table_4096_fc32_size,
};

```

I think it should be OK. Can you please try?

Thanks,
Dmitry

ashvarma-git, 2020-12-16:
Yes, of course that fixes the compiler error in this file.
Then the dotprod example fails compile at this file next, due to the same issue:

modules/fft/float/dsps_fft4r_bitrev_tables_fc32.c
Please review for all such errors.

dmitry1945, 2020-12-18:
@ashvarma-git 
changes in master.
Please pull.

Thank you very much!

Dmitry

closed 2022-03-07 by igrr