espressif/esp-dsp #23: FFT4Real Example Error

Vvard, 2021-02-17:
I've tested fft4real example for 2048, 1024, 512, 256, 128, 64 samples, but fft4r function for real input only works for 2048, 512, 128 cases.

[image]
[image]
[image]
[image]
[image]
[image]

dmitry1945, 2021-02-17:
Hi @Vvard ,

The result is correct. 
In example we are using FFT radix-2 and radix-4. 
The real N samples converted to N/2 complex samples. Then  samples processed by FFT and then converted back to real spectrum. 
If you use 1024 input samples, then 512 complex samples could not be processed by FFT radix-4, and will be processed only by radix-2.
And you see correct result for radix-2 and  wrong result for radix-4.

Regards,
Dmitry

Vvard, 2021-02-17:
Hi @dmitry1945 ,

Thank you for pointing the error.
And  it would be nice if fft4r would return an error, that specific numbers of samples are not supported.

Regards,
Vard

closed 2021-03-04 by Vvard