espressif/esp-dsp #24: Compilation error in fft4real example for ANSI DSP optimization configuration.

Vvard, 2021-02-17:
I tested fft4real example for ANSI DSP optimization but got below compile error.

[image]

Error:
C:/esp-dsp/modules/fft/include/dsps_fft4r.h:157:29: error: implicit declaration of function 'dsps_bit_rev4r_fc32_ansi'; did you mean 'dsps_bit_rev_fc32_ansi'? [-Werror=implicit-function-declaration]

Vvard, 2021-02-23:
Hi team,
Just wanted to know if you had a look into this issue?

Additional info: There is no dsps_fft4r_fc32_ansi function declaration in the code
https://github.com/espressif/esp-dsp/blob/master/modules/fft/include/dsps_fft4r.h#L168

dmitry1945, 2021-02-23:
Hi @Vvard 
I will fix it today.
The new version will be available soon.

Thank you very much,
Dmitry

Vvard, 2021-02-23:
Hi @dmitry1945,
Thanks a lot, I will wait for the new version)

Regards,
Vard

closed 2021-03-04 by Vvard