espressif/esp-dsp #31: ESP-DSP optimized functions can't be found during linking(IDFGH-5484)

ahmedwahdan, 2021-07-14:
## Environment

- Development Kit:     ESP32-DevKitC
Module or chip used:  ESP32-WROOM-32
- IDF version  4.2
- Build System:         idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     WindowsPlain Command Prompt
- Using an IDE?: No|
- Power Supply:         USB

## Problem Description

After adding ESP-DSP component to esp-idf components, only ANSI C function can be compiled, the optimized functions couldn't be found

[image]

### Expected Behavior
Be able to use optimized DSP functions
### Steps to reproduce

1. Clone DSP component to idf components
2. configure the DSP to use optimized functions

[image]

3. Build any DSP example

Build output:
[build_output.txt](https://github.com/espressif/esp-idf/files/6758288/build_output.txt)

igrr, 2021-07-14:
Sorry for the late response @ahmedwahdan, it is a bug we introduced in https://github.com/espressif/esp-dsp/commit/901d03f4f92e0661564598f3b9e939c678481d7c. CONFIG_IDF_TARGET_ARCH_XTENSA is not defined in IDF 4.2.x, and was only added in IDF 4.3.

ahmedwahdan, 2021-07-14:
No problem, What about the V3.3, it seems that the sdkconfig.h is included from different path which doesn't include CONFIG_IDF_TARGET_ARCH_XTENSA

igrr, 2021-07-14:
Yes, it's the same for v3.3, CONFIG_IDF_TARGET_ARCH_XTENSA is not defined there.
This issue in esp-dsp happens with every IDF version older than v4.3.

closed 2021-07-15 by espressif-bot