espressif/esp-dsp #33: Matrix subtraction does not work correctly

chibi314, 2021-09-16:
Hi.

I found a serious problem on matrix subtraction.
The subtraction function returns the opposite sign matrix.
In other words, when you calculate ```A - B```, the function returns ```B - A```.

Please try [this](
https://gist.github.com/chibi314/f3cd2ebcf3406016f1f30b3ad0eb0f52#file-dspm_matrix_main-cpp) code replacing the matrix example [here](https://github.com/espressif/esp-dsp/blob/master/examples/matrix/main/dspm_matrix_main.cpp).

In addition, I try this code on M5Stack Fire.

How could we fix this problem?

chibi314, 2021-09-16:
Hi.
Sorry to send a post again.

I just fixed [this line](https://github.com/espressif/esp-dsp/blob/master/modules/math/sub/float/dsps_sub_f32_ae32.S#L54) to

```
        sub.s    f2, f0, f1     // f2 = f0 - f1
```
and it works well now.

Can I send a PR about this?

dmitry1945, 2021-09-16:
Hi @chibi314 

Thank you for the fix. Yes, you can make an PR, or I can make it. It's up to you.

Thanks,
Dmitry

chibi314, 2021-09-17:
@dmitry1945 

Hi.
I sent a PR #34 .

I'm grateful for this good library.
I'm sorry to forgot to say this at first.

Thanks.

closed 2021-11-18 by chibi314