espressif/esp-dsp #41: esp-dsp library question

darwin1234, 2022-03-23:
These marcros are from CIMSIS-DSP
arm_cfft_sR_f32_len256
arm_cfft_instance_f32
arm_cfft_f32

My question is what is the equivalent of those macros with esp-dsp library?

Thanks,

----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

dmitry1945, 2022-03-23:
Hi @darwin1234 ,
In CIMSIS-DSP you have to initialize FFT for every instance. In Esp-DSP you have to initialize FFT once for all instances. That's why we have no full analog of arm_cfft_instance_f32 and arm_cfft_sR_f32_len256. And arm_cfft_f32 is not one to one.
In Esp-DSP you have to initialize FFT module once by calling:

dsps_fft2r_init_fc32(...)
then, depends on what you plan to do, you can use:
dsps_fft2r_fc32() and dsps_bit_rev_fc32(...)
or you can use just dsps_fft2r_fc32().
That's it.
You can look to the example here:
https://github.com/espressif/esp-dsp/tree/master/examples/fft4real
or here:
https://github.com/espressif/esp-dsp/tree/master/examples/fft

darwin1234, 2022-03-23:
Hi @dmitry1945,

Thanks for the quick response! :)  I modified the code.  Check below.

This is the original code that uses CIMSIS-DSP library

`	const static arm_cfft_instance_f32* maskS;  ///<----
	maskS = &arm_cfft_sR_f32_len256; ///<----
	int ptr;
	
	for (unsigned j = 0; j < np; j++)     
	{
		memset(maskgen, 0, partitionsize * 16);  // zero out maskgen
		// take part of impulse response and fill into maskgen
		for (unsigned i = 0; i < partitionsize; i++)
		{
			// THIS IS FOR REAL IMPULSE RESPONSES OR FIR COEFFICIENTS
			// the position of the impulse response coeffs (right or left aligned)
			// determines the useable part of the audio in the overlap-and-save (left or right part of the iFFT buffer)
			ptr = i + j * partitionsize;
			if (ptr < size) {
				maskgen[i * 2 + partitionsize * 2] = guitar_cabinet_impulse[ptr];
			}
			else  // Impulse is smaller than the fixed (defined) filter length (taps)
			{
				maskgen[i * 2 + partitionsize * 2] = 0.0;  // pad the last part of filter mask with zeros
			} 
		}
			// perform complex FFT on maskgen
		arm_cfft_f32(maskS, maskgen, 0, 1);  ///<----
		// fill into fmask array
		for (unsigned i = 0; i < partitionsize * 4; i++)
		{
			fmask[j][i] = maskgen[i];
		}
	}
	enabled = 1;
	Serial.println("Mask file generated");`

----------------------------------------------------
and this is ESP-DSP library

`float maskgen[N_SAMPLES*2];
	
	ret = dsps_fft2r_init_fc32(NULL, CONFIG_DSP_MAX_FFT_SIZE);
	int ptr;
	
      if (ret  != ESP_OK)
      {
          ESP_LOGE(TAG, "Not possible to initialize FFT. Error = %i", ret);
         return;
      }
	
     for (unsigned j = 0; j < np; j++)     
     {

		for (unsigned i = 0; i < partitionsize; i++)
		{
			// THIS IS FOR REAL IMPULSE RESPONSES OR FIR COEFFICIENTS
			// the position of the impulse response coeffs (right or left aligned)
			// determines the useable part of the audio in the overlap-and-save (left or right part of the iFFT buffer)
			ptr = i + j * partitionsize;
			if (ptr < size) {
				maskgen[i * 2 + partitionsize * 2] = g_cabinet[ptr];
			}
			else  // Impulse is smaller than the fixed (defined) filter length (taps)
			{
				maskgen[i * 2 + partitionsize * 2] = 0.0;  // pad the last part of filter mask with zeros
			} 
		}
		
		dsps_fft2r_fc32(fmask, maskgen);
		
		for (unsigned i = 0; i < partitionsize * 4; i++)
		{
			fmask[j][i] = maskgen[i];
		}
	}
	enabled = 1;`

Is this correct ?

Thanks,

dmitry1945, 2022-03-23:
Hi @darwin1234 
I htink you have forgoten to call dsps_bit_rev_fc32() after dsps_fft2r_fc32();

I will look somethink like this:

dsps_fft2r_fc32(maskgen, N_SAMPLES);
dsps_bit_rev_fc32(maskgen, N_SAMPLES);

...

closed 2023-07-28 by dmitry1945