espressif/esp-dsp #49: FFT example does not generate proper results with IDF Master

davidallenmann, 2022-07-21:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-S3-DEVKITC N8R8
- Kit version DevKitC
- Module or chip used:  ESP32-S3-WROOM1
- IDF version: v5.0-dev-4303-g075e0729de
- Build System:         CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     macOS
- Power Supply:         USB

## Problem Description

FFT example builds and runs on ESP32-S3 using latest V5 IDF, but it generates erroneous output. No changes were made to this example: https://github.com/espressif/esp-dsp/tree/master/examples/fft

### Expected Behavior
Plots like shown in readme with single or double peak.
```
I (59) main: Start Example.
W (89) main: Signal x1
I (89) view: Data min[495] = -162.760925, Data max[164] = 23.938747
 ________________________________________________________________
0                                                                |
1                    |                                           |
2                    |                                           |
3                    |                                           |
4                    |                                           |
5                    |                                           |
6                   | |                                          |
7                   | |                                          |
8                  || ||                                         |
9||||||||||||||||||     ||||||||||||||||||||||||||||||||||||||||||
 0123456789012345678901234567890123456789012345678901234567890123
I (159) view: Plot: Length=512, min=-60.000000, max=40.000000
W (169) main: Signal x2
I (169) view: Data min[502] = -164.545135, Data max[205] = 3.857752
 ________________________________________________________________
0                                                                |
1                                                                |
2                                                                |
3                         |                                      |
4                         |                                      |
5                         |                                      |
6                         |                                      |
7                         ||                                     |
8                        | |                                     |
9||||||||||||||||||||||||   ||||||||||||||||||||||||||||||||||||||
 0123456789012345678901234567890123456789012345678901234567890123
I (249) view: Plot: Length=512, min=-60.000000, max=40.000000
W (249) main: Signals x1 and x2 on one plot
I (259) view: Data min[505] = -159.215271, Data max[164] = 23.938747
 ________________________________________________________________
0                                                                |
1                    |                                           |
2                    |                                           |
3                    |    |                                      |
4                    |    |                                      |
5                    |    |                                      |
6                   | |   |                                      |
7                   | |   ||                                     |
8                  || || | |                                     |
9||||||||||||||||||     |   ||||||||||||||||||||||||||||||||||||||
 0123456789012345678901234567890123456789012345678901234567890123
I (339) view: Plot: Length=512, min=-60.000000, max=40.000000
I (339) main: FFT for 1024 complex points take 140472 cycles
I (349) main: End Example.

```

### Actual Behavior
```
W (349) main: Signal x1
I (349) view: Data min[151] = -40.703781, Data max[1] = 10.201058
 ________________________________________________________________
0                                                                |
1                                                                |
2| |                                                          | ||
3| |||   | |||| || ||||||||| ||||  | || | || || ||   |||||   |||||
4||||||||||||||||||||||||||||||||||||||||||||||||| ||||||||||||| |
5 |  |||||| | ||| |||  ||   |   |||||| ||| || ||| ||| |||||||    |
6 |                |        |     |      |        |||            |
7                                                                |
8                                                                |
9                                                                |
 0123456789012345678901234567890123456789012345678901234567890123
I (419) view: Plot: Length=512, min=-60.000000, max=40.000000
W (429) main: Signal x2
I (429) view: Data min[272] = -52.645767, Data max[1] = 10.407180
 ________________________________________________________________
0                                                                |
1                                                                |
2| |                                                          | ||
3| |||   | ||||||| ||||||||  |||| |  |||||||  | |    |||||   |||||
4||||| ||| ||||||||||||||||||||||||||||||||||||||| ||||||||||||| |
5 | ||||| ||| |||||     |  |    || ||  | |   |||||||| | |||||    |
6 |   ||       |  |                |   |       |  ||        |    |
7              |                   |                        |    |
8                                  |                             |
9                                                                |
 0123456789012345678901234567890123456789012345678901234567890123
I (509) view: Plot: Length=512, min=-60.000000, max=40.000000
W (519) main: Signals x1 and x2 on one plot
I (519) view: Data min[405] = -31.928421, Data max[1] = 10.407180
 ________________________________________________________________
0                                                                |
1                                                                |
2| |                                                          | ||
3| |||   | ||||||| ||||||||| |||| || ||||||| || ||   |||||   |||||
4 |||||||||||||||||||||||||||||||||||||||||||||||| ||||||||||||| |
5 | | |||     ||| |     |       |   |    |    ||  ||| |  ||||    |
6                                                 |              |
7                                                                |
8                                                                |
9                                                                |
 0123456789012345678901234567890123456789012345678901234567890123
I (599) view: Plot: Length=512, min=-60.000000, max=40.000000
I (609) main: FFT for 1024 complex points take 130110 cycles
I (609) main: End Example.

```

davidallenmann, 2022-07-21:
I went back to v4.4.
I got a similar result when I built without configuring PSRAM and clock speed. I changed the clock speed to 240 MHz and configured Octal SPIRAM, and the example now works correctly.

davidallenmann, 2022-07-21:
Switched back to v5.0
Configured PSRAM and changed clock speed to 240 MHz.
Still get same result with bad output.

dmitry1945, 2022-07-22:
Hi @davidallenmann ,

I will look today and will try to reproduce.
 
Thanks for report.

dmitry1945, 2022-07-22:
@davidallenmann 
The problem was aligment allocation of table for FFT.
I have fix the error in local repo and will soon push it to github, but if you need changes now please do follows in file
In file modules/fft/float/dsps_fft2r_fc32_ansi.c:
1. Add to the list of headers:
#include <malloc.h>
2. Exchange line 66 from:
`                    dsps_fft_w_table_fc32 = (float *)malloc(table_size * sizeof(float));
`
to 
`                    dsps_fft_w_table_fc32 = (float*)memalign(16, sizeof(float) * table_size);
`
That will solve your problem.

Regards,
Dmitry

davidallenmann, 2022-07-22:
Thanks for the fast response. That fixes it!

closed 2022-07-22 by davidallenmann