espressif/esp-dsp #50: modules/math/sqrt/float/dsps_sqrt_f3_ansi.c build error

davidallenmann, 2022-07-23:
## Environment

- Development Kit:     ESP32-S3-DevKitC
- IDF version: v5.0-dev-4303-g075e0729de
- Build System:         CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:    macOS
- Power Supply: USB

## Problem Description
Build doesn't error if build with compiler debug mode, but when build with compiler optimization 'Optimize for performance (-O2)' get build error as below.

```
/Users/davidmann/w/Haikubox2-firmware/modules/math/sqrt/float/dsps_sqrt_f32_ansi.c:24:12: error: 'result' is used uninitialized [-Werror=uninitialized]
   24 |     return *f_result;
      |            ^~~~~~~~~
/Users/davidmann/w/Haikubox2-firmware/modules/math/sqrt/float/dsps_sqrt_f32_ansi.c:22:15: note: 'result' declared here
   22 |     const int result = 0x1fbb4000 + (*f_ptr >> 1);

```

Should it be as follows to return a float instead of a pointer to a float?
```
inline float dsps_sqrtf_f32_ansi(float f)
{
    int* f_ptr = (int*)&f;
    const int result = 0x1fbb4000 + (*f_ptr >> 1);
    float* f_result = (float*)&result;
    return f_result;   
}
```

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

dmitry1945, 2022-07-24:
Hi @davidallenmann ,
Yes, this is the problem with latest GCC. We will fix it soon.

Thanks,
Dmitry

dmitry1945, 2022-07-25:
Hi @davidallenmann 
The update will be soon.
Right now please change `dsps_sqrtf_f32_ansi` to

```
inline float dsps_sqrtf_f32_ansi(float f)
{
    int result;
    int* f_ptr = (int*)&f;
    result = 0x1fbb4000 + (*f_ptr >> 1);
    const int *p = &result;
    float* f_result = (float*)p;
    return *f_result;   
}

```

Barabas5532, 2022-12-05:
This is fixed in https://github.com/espressif/esp-dsp/commit/7a13333127f29a24a2c49d785cdbbe772bdd036b

closed 2023-07-28 by dmitry1945