espressif/esp-dsp #51: Compile of examples fail

eyewy, 2022-08-28:
[ 89%] Built target __idf_wifi_provisioning
In file included from /esp-dsp/modules/support/view/dsps_view.cpp:3:
/esp-dsp/modules/support/view/dsps_view.cpp: In function 'void dsps_view(const float*, int32_t, int, int, float, float, char)':
/esp-idf/components/log/include/esp_log.h:265:27: error: format '%i' expects argument of type 'int', but argument 6 has type 'int32_t' {aka 'long int'} [-Werror=format=]
  265 | #define LOG_COLOR(COLOR)  "\033[0;" COLOR "m"

dmitry1945, 2022-08-29:
Hi @eyewy ,

Please pull the lates master. The problem should be solved.

Thank you.

eyewy, 2022-08-30:
I still have other errors, which branch of esp-idf shall I use? Master OK?

dmitry1945, 2022-08-30:
Hi @eyewy,
yes, it should be master. And lates IDF.
Could you show the errors please?

Dmitry

eyewy, 2022-08-31:
CMake Error at /esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 51

  

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    /esp-idf/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  

  CMake Error at /esp-idf/CMakeLists.txt:2
  (project):

    project command is not scriptable

  Call Stack (most recent call first):

    /esp-idf/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    /esp-idf/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  /esp-idf/tools/cmake/build.cmake:553 (__component_get_requirements)
  /esp-idf/tools/cmake/project.cmake:434 (idf_build_process)
  CMakeLists.txt:9 (project)

-- Configuring incomplete, errors occurred!

eyewy, 2022-08-31:
I get the above error with the "kalman" example. Tried compiling "fft" and it went OK.

dmitry1945, 2022-08-31:
@eyewy 

Kalman example should work now.

Thank you, @eyewy , for the feedback.

Regards,
Dmitry

eyewy, 2022-09-01:
Thanks, closing as issue resolved.

closed 2022-09-01 by eyewy