espressif/esp-dsp #52: Resampler in esp-dsp, feature request

AristarchosOfSamos, 2022-08-31:
Hi,
There are many occasions where a resampler is needed without having the complete esp-adf framework.
Specially since esp-adf is relying in many tweaks and specific versions or git checkouts that makes it almost impossible to properly use it or even try to follow its own intricacies. All these make it a really difficult environment to use.
A very handy function that everybody in DSP needs is the resampler but unfortunately it is inside esp-adf..

In my view, resampler belongs better in the esp-dsp,  whereas almost all such DSP libraries got a resample function as this is the norm.
Please consider to add it in the esp-dsp since the esp-sdp can be actually used with almost all esp-idf versions, even the master/development.

dmitry1945, 2022-08-31:
@AristarchosOfSamos,

Thank you for feedback.
We will add resampler to esp-dsp. I'm not promise the exact day, but I hope, it will be this year.

Regards,
Dmitry

dmitry1945, 2025-06-17:
Hi @AristarchosOfSamos ,

The sample rate converter now in esp-dsp. You can use it.

closed 2025-09-01 by Indastri