espressif/esp-dsp #55: Is dsps_fir_f32_aes3 implementation working?

AristarchosOfSamos, 2022-09-28:
Hi,

Trying to utilize FIR on a ESP32S3 (ESP32-S3-DevKitC-1) and seen the esp-dsp FIR implementation file here
[https://github.com/espressif/esp-dsp/blob/master/modules/fir/float/dsps_fir_f32_aes3.S](https://github.com/espressif/esp-dsp/blob/master/modules/fir/float/dsps_fir_f32_aes3.S)

Tried a test of my own (based on test_app) and its functionality show FAIL whereas its result is different from the standard ansi result.

Is there any example/test that we can do for ESP32S3 on esp-dsr FIR?
Is this implementation file working correctly?

Regards,

dmitry1945, 2022-09-28:
Hi @AristarchosOfSamos ,
FIR for floating point is not working now. 
Will fix soon.
For now, please use esp32 or ansi version.

Regards,
Dmitry

herculanodavi, 2023-03-01:
Hello,

do we have any news on this functionality?

dmitry1945, 2023-03-01:
Hi @herculanodavi 
I'ts in progress.
Will come soon.

dmitry1945, 2023-07-28:
@herculanodavi  it's working. I will close the issue.

Thank you very much,
Dmitry

closed 2023-07-28 by dmitry1945