espressif/esp-dsp #56: dsps_fft2r_sc16_ansi_ should malloc aligned memory?

djfurie, 2022-10-21:
https://github.com/espressif/esp-dsp/blob/401faf8bcf76bb5c958c83f39cb85627fdacc156/modules/fft/fixed/dsps_fft2r_sc16_ansi.c#L92

For use with function `dsps_fft2r_sc16_aes3` should the init function be using 16 byte aligned memory?

dmitry1945, 2022-10-22:
Hi @djfurie ,
I think yes,, iit should use the memalign(...)

I will fix it soon.

Thank you very much!
Dmitry

ayavilevich, 2023-01-31:
I consider this a critical bug. This will "randomly" break ``dsps_fft2r_sc16`` = ``dsps_fft2r_sc16_aes3`` depending on your memory map and heap arrangement. The function will run but will return invalid data.

A work around for now is to use a static buffer for the init. For example:

```
__attribute__((aligned(16))) int16_t dsps_fft2r_init_sc16_static_buf[CONFIG_DSP_MAX_FFT_SIZE];
dsps_fft2r_init_sc16(dsps_fft2r_init_sc16_static_buf, CONFIG_DSP_MAX_FFT_SIZE);
```

dmitry1945, 2023-07-28:
@djfurie the issue is fixed in 1.4.x version.

Thank you very much!
Dmitry

closed 2023-07-28 by dmitry1945