espressif/esp-dsp #60: Inverse of dsps_cplx2real_fc32

ayavilevich, 2022-12-11:
Request for a feature or assistance.

I learned that an optimal fft of a N real input can be done using a full fft on an N/2 input and then a call to dsps_cplx2real_fc32 .
I understand that an fft of a complex input/output can be used as an ifft with some small adjustments.

I am looking to do an efficient ifft where the output is expected to be real and the input is the same format that dsps_cplx2real_fc32 returns.
Is it possible to do it in an efficient way like it is possible with the forward direction? Do we need an inverse of dsps_cplx2real_fc32 and then a call to a full fft with N/2 values?
Please advise.

dmitry1945, 2023-08-15:
Hi @ayavilevich ,

the forward FFT is complex conjuction of inverce FFT. You can just make FFT that we have, calculate the result and and then multiply imag part to -1.

Regards,
Dmitry

closed 2023-08-21 by dmitry1945

coderforlife, 2025-07-24:
I have discovered the underlying issues for reversing `dsps_cplx2real_fc32()`. It requires special handling of the 0 element in the array. See the code at https://github.com/espressif/esp-dsp/issues/25#issuecomment-3112378403 for a solution.