espressif/esp-dsp #62: Support iFFT

howard0su, 2023-01-15:
Current dsp lib supports only FFT, please add iFFT support. It is common to process signals in frequency domain which do fft, filter, ifft.

dmitry1945, 2023-01-18:
Hi @howard0su 

Yes, we will add.

Regards,
Dmitry

drewbharris, 2024-03-06:
@dmitry1945 any updates on this? Thanks!

copych, 2024-07-26:
I am not that good at math, but having FFT working it shouldn't be that hard to implement iFFT. But if we do that outside the ESP_DSP, that'll be waist of CPU cycles, cause we need to swap Re and Im arrays twice, while inside the library code you just change indexing instead. So please, add iFFT.

crosswick, 2024-11-02:
I am also very interested in efficient iFFT.

coderforlife, 2025-07-24:
I have a working solution at  https://github.com/espressif/esp-dsp/issues/25#issuecomment-3112378403 that does not modify the ESP-DSP code at all. It does require two additional (quick) passes through the data to perform complex conjugation. If additional functions were made in ESP-DSP this could be done with several sign changes in the code instead of needing additional passes.