espressif/esp-dsp #64: Use of ESP-DSP with ESP-ADF

ayavilevich, 2023-01-31:
Hi, I noticed that ESP-ADF contains ESP-SR which in turn contains a pre-compiled version of the ESP-DSP but for S3 only.
``esp-adf\components\esp-sr\lib\esp32s3\libesp-dsp.a``

Is this by design?

I have a project with ESP-DSP and ESP-ADF together and it seems to link with the precompiled library not the sources of the ESP-DSP repository.
Can this lead to mismatch between the includes and the library file?
Can this be controlled?
Please provide best practices.

dmitry1945, 2023-01-31:
Hi @ayavilevich 

Thank you very much for this important issue.
I will look.

feizi, 2023-02-01:
> Is this by design?

> Hi, I noticed that ESP-ADF contains ESP-SR which in turn contains a pre-compiled version of the ESP-DSP but for S3 only. `esp-adf\components\esp-sr\lib\esp32s3\libesp-dsp.a`
> 
> Is this by design?

Yes, libesp-dsp.a in ESP-SR is a pre-compiled version of the ESP-DSP for S3.
I can separate ESP-DSP by component manager from ESP-SR, but I think the best way is that user can control which library to link.

ayavilevich, 2023-02-01:
> Yes, libesp-dsp.a in ESP-SR is a pre-compiled version of the ESP-DSP for S3. I can separate ESP-DSP by component manager from ESP-SR, but I think the best way is that user can control which library to link.

Having user control over it sounds like a good option. Is this something that a user can already do today or something that needs to be added by Espressif?

igrr, 2023-02-14:
esp-sr has been updated to use esp-dsp as a component in https://github.com/espressif/esp-sr/commit/e1c73a59d94fe5d656c83788c9cec619e8f5070a. Some extra work might be required on the ADF side to update to the newer esp-sr release.