espressif/esp-dsp #68: SOLVED: A 1½cycle/tap fp32 FIR filter with arbitrary length and decimation

f4lc0n-asm, 2023-02-23:
Hello,

this one was a tough nut to crack, but ultimately was successful! :)

Notable differences:
• Coefficients and delay line must begin and end at **16-byte boundary**. Use **__attribute__((aligned (16)))** and **SIZE_ALIGN16()** macro to achieve this. Both are required because of 128-bit reads from memory.
• Coefficients must be **reversed** unless symmetrical (quite often). This is to achieve maximum speed by keeping the ASM code much simpler.
• **uint** is used instead of **int** in **fir_f32_s** struct because these values are always non-negative.
• Since the non-decimating FIR filter doesn't produce any errors, the output is declared **void**.

I am attaching the sources with tests. Use **7-Zip** to extract them from the ZIP archive. Feel free to incorporate them into this project. Let me know if you need the ASM sources tweaked by me.

Cheers!

f4lc0n
[FIR_fp32_1.5c_per_tap.zip](https://github.com/espressif/esp-dsp/files/10815864/FIR_fp32_1.5c_per_tap.zip)

dmitry1945, 2023-02-24:
HI @f4lc0n-asm 
I will look, and will include your part.

Thank you very much!

f4lc0n-asm, 2023-03-01:
Hello @dmitry1945,

the speed optimizations are a strange thing - they just pop out in my mind out of the blue, I don't have to consciously think about them, mull them over and over. But let us consider the latest **v1.5** to be **final**, at least for the time being. The artist in me wants to move on…

Since I haven't heard any screams on the internet that the FIR filters don't produce the correct output, etc., and also they pass my internal tests - it is a wrap! :)

Should you need any assistance from me, just ask here…

Kind regards,

f4lc0n

f4lc0n-asm, 2023-03-03:
Hello,

since I heard some people screaming that FIR coefficients must be reversed, I prepared a new version, which accepts coefficients in normal order. However, the delay line is now in reversed order! Most people will just shrug it off. The FIR performance is very similar to the reversed coefficients version.

**Notable differences:**
• Coefficients and delay line must begin and end at 16-byte boundary. Use **ALIGN16** and **SIZE_ALIGN16()** macros to achieve this. Both are required because of 128-bit reads from memory.
• **uint** is used instead of **int** in **fir_f32_s** struct because these values are always non-negative (ASM routines use unsigned compare).
• Since the non-decimating FIR filter doesn't emit any errors, the function output is declared **void**.

I am attaching the sources with tests. Use **7-Zip** to extract them from the 7-Zip archive. Feel free to incorporate them into this project. Let me know if you need me to tweak the ASM sources.

Cheers!

f4lc0n

[FIR_fp32_1.5c_per_tap_v2.0.zip](https://github.com/espressif/esp-dsp/files/10876783/FIR_fp32_1.5c_per_tap_v2.0.zip)

dmitry1945, 2023-07-25:
Hi @f4lc0n-asm the FIR filters optimyzed for esp32s3 now in the esp-dsp. The implementation is different that you have placed, but the interface and requirements for aligment are the same. I think you will be able to use it in your project without modifications.

Please take a look.

That you, and best regards,
Dmitry

f4lc0n-asm, 2023-07-25:
Hello Dmitry1945,

just looking at **esp-dsp/blob/master/modules/fir/float/dsps_fir_init_f32.c line 44:** `if (fir->N%4 != 0)` - does it mean that FIR length **N must be an integer multiple of 4** in your ESP32-S3 FIR solution? I devised solution just for **any FIR length**, even prime numbers for mathematics afficionados :)

f4lc0n

dmitry1945, 2023-07-25:
Hi @f4lc0n-asm 
The parameter len, yes,  we have a trick. It should be divided by 4, but, you can use any length, just add zero the the end of the impulse response.

f4lc0n-asm, 2023-07-25:
Just looking at your FIR ASM sources. Some optimization tips (line numbers are from dsps_fir_f32_aes3.S, but it the same spirit apply to dsps_fird_f32_aes3.S):

- During the FIR computation, the delay line is read from the oldest sample to the newest - the coefficients must reflect that, i.e. first padding with float zeros, if any, then the FIR coefficients in reverse order - you can easily avoid that by storing the samples into the delay line with **decrementing pos** like I do in my FIR routines.
- **AND** instruction at line 76 is superfluous because **EE.LDF.128** instruction automatically and internally zeroes out 4 address LSBs before reading the 4 floats - **A12** and **A15** are no longer needed.
- Second loops for offsets 1, 2, and 3 can be optimized. E.g. for offset 1 delete the instruction at line 138 and move the instruction from line 146 inbtwn the lines 140 and 141 (it becomes the 1st instruction in the loop instead of the last one).
- **AND** instruction at line 86 can be replaced with **EXTUI a8, a11, 2, 2** - **A13** is no longer needed. You will have bits 3 and 2 shifted into bits 1 and 0. Test them for zero after decrementing with **ADDI.N a8, a8, -1**.

Well, 3 solid solutions appeared! The important thing is that people have a choice! As for my choice - I only trust my ASM code so much that I even create custom RTOS in ASM for my applications. So I guess you already know the answer! ;)

Cheers!

f4lc0n

closed 2023-07-28 by dmitry1945