espressif/esp-dsp #70: Wrong output when running dsps_mulc_s16_ functions

gui88Abreu, 2023-06-02:
## Environment

I am running this library from the latest arduino-esp32 package release (2.0.9). Since I'm assuming the issue is not directly related to arduino-esp32 package, I'm opening the issue here. I'm sorry if this is a mistake. 

I'm compiling the project using the latest arduino-cli release as well.

- Development Kit:      [none]
- Module or chip used:  [ESP32S3]
- Arduino-cli: [0.33.0]
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

I was trying to use the dsps_mulc functions to multiply an array by a constant, as described in its documentation, but I realized that the output for the int16_t version of the function is either [-1,-1,-1,...] or [0,0,0,...]. However, the float version works very well, always providing the expected output.

Example:

```cpp
#include <esp_dsp.h>

const int16_t input[4] = {5,2,8,9};
int16_t output[4];
const int16_t myConstant = -5;
esp_err_t dsp_error;

dsp_error = dsps_mulc_s16_ae32(input, output, 4, -5, 1, 1);
//same happens with the ansi version;
```

### Expected Behavior

dsp_error = 0;
Output: [-25, -10, -40, -45]

### Actual Behavior

dsp_error = 0;
Output: [-1, -1, -1, -1]

### Steps to reproduce

Compiling with the following options and executing the above piece of code should reproduce the bug.

_id=esp32s3
bootloader.tool=esptool_py
bootloader.tool.default=esptool_py
build.arch=ESP32
build.board=ESP32S3_DEV
build.boot=qio
build.boot_freq=80m
build.bootloader_addr=0x0
build.cdc_on_boot=1
build.code_debug=4
build.copy_jtag_files=0
build.core=esp32
build.core.path=
build.custom_bootloader=bootloader
build.custom_partitions=partitions
build.defines=
build.dfu_on_boot=0
build.event_core=-DARDUINO_EVENT_RUNNING_CORE=1
build.extra_flags=-DESP32 -DCORE_DEBUG_LEVEL=4 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1  -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0
build.extra_flags.esp32=-DARDUINO_USB_CDC_ON_BOOT=0
build.extra_flags.esp32c3=-DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1
build.extra_flags.esp32s2=-DARDUINO_USB_MODE=0 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0
build.extra_flags.esp32s3=-DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0
build.extra_libs=
build.f_cpu=240000000L
build.flash_freq=80m
build.flash_mode=dio
build.flash_size=8MB
build.fqbn=esp32:esp32:esp32s3:PartitionScheme=8MBCustom,FlashMode=qio,EraseFlash=all,CDCOnBoot=cdc,UploadMode=cdc
build.library_discovery_phase=0
build.loop_core=-DARDUINO_RUNNING_CORE=1
build.mcu=esp32s3
build.memory_type=qio_qspi
build.msc_on_boot=0
build.openocdscript=esp32s3-builtin.cfg
build.openocdscript.esp32=esp32-wrover-kit-3.3v.cfg
build.openocdscript.esp32c3=esp32c3-builtin.cfg
build.openocdscript.esp32s2=esp32s2-kaluga-1.cfg
build.openocdscript.esp32s3=esp32s3-builtin.cfg
build.opt.name=build_opt.h
build.opt.path=
build.partitions=8Mpartition
build.path=
build.project_name=test_bench.ino
build.psram_type=qspi
build.source.path=
build.system.path=
build.tarch=xtensa
build.target=esp32s3
build.usb_mode=1
build.variant=esp32s3
build.variant.path=

The code was compiled using the following command line:
arduino-cli compile -b esp32:esp32:esp32s3 <source-path>  --board-options "PartitionScheme=8MBCustom,FlashMode=qio,EraseFlash=all,CDCOnBoot=cdc,UploadMode=cdc"  --build-property "compiler.optimization_flags=-O2" --build-property "build.flash_size=8MB" --build-property "build.partitions=8Mpartition" --build-property "build.code_debug=4" --output-dir source/build/bpi-board/output --log-format text --log-file source/build/temp/build.log --log-level warn --verbose

## Other items

[elf-file.zip](https://github.com/espressif/esp-dsp/files/11638902/elf-file.zip)

peter-marcisovsky, 2023-06-05:
Hello, the problem here is data format. In the esp-dsp repository, all the `_s16_` functions use Q15 fixed point data format. In your code, you are feeding the function with incorrect inputs, which causes an output overflow, and this is the reason why you are getting `-1` output for your input. 

Peter

guilhAbreu, 2023-06-06:
@peter-marcisovsky Thank you for your time, and I'm sorry to continue to bother you. I took a look at both ANSI and ae32 implementations, and simply just removing the shift instruction solved the problem. It would be very helpful if the shift parameter could be implemented.

For those who are facing the same issue. I'm talking about the lines 41 and 24 of the ae32 and ANSI implementation, respectively. I'm appending here the custom implementations that solve the issue.
[solution.zip](https://github.com/espressif/esp-dsp/files/11671127/solution.zip)

closed 2023-06-09 by guilhAbreu