espressif/esp-dsp #74: Two environments with different actions were specified for the same target

ProjectRobal, 2023-08-16:
I am trying to compile PlatformIO project with ESP-DSP on ESP32-S3 and I am getting error during compilation:

*** Two environments with different actions were specified for the same target: /home/projectrobal/Documents/PlatformIO/Projects/GaraiCPU/.pio/build/esp32-s3/components/esp-dsp/modules/support/cplx_gen/dsps_cplx_gen.o
(action 1: xtensa-esp32s3-elf-gcc -g2 -ggdb2 -DPLATFORMIO=60110 -DARDUINO_ESP32S3_DEV -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1 -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -D_GNU_SOURCE -DIDF_VER=\"5.0.2\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -D__PLATFORMIO_BUILD_DEBUG__ -c -o dsps_cplx_gen.o dsps_cplx_gen.S)
(action 2: xtensa-esp32s3-elf-gcc -mlongcalls -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=/home/projectrobal/Documents/PlatformIO/Projects/GaraiCPU=. -fmacro-prefix-map=/home/projectrobal/.platformio/packages/framework-espidf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -g2 -ggdb2 -DPLATFORMIO=60110 -DARDUINO_ESP32S3_DEV -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1 -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -D_GNU_SOURCE -DIDF_VER=\"5.0.2\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -D__PLATFORMIO_BUILD_DEBUG__ -c -o dsps_cplx_gen.o dsps_cplx_gen.S)
File "/home/projectrobal/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 684, in compile_source_files

My platform.ini file:
`[env:esp32-s3]
platform = espressif32 @^6.3.2
board = esp32-s3-devkitc-1-n16r8v
framework = espidf
monitor_speed=115200
monitor_filters=esp32_exception_decoder
build_type = debug
board_build.partitions=./partition.csv`

I don't what could cause it and how to resolve it. I have git cloned esp-dsp into components folder of my project.

Ackites, 2023-08-25:
I have the same problem.Is there any solution?

dmitry1945, 2023-08-25:
Hi @ProjectRobal  @Ackites ,
I will look today. Will inform you when solution will be ready.

Thank you

dmitry1945, 2023-08-25:
@ProjectRobal 

could you please guide me, how you build your project, and which project are you building?

Thanks,
Dmitry

ProjectRobal, 2023-08-25:
If that would help I have made changes to platform.ini file and it works now: 

`; PlatformIO Project Configuration File 
 ; 
 ;   Build options: build flags, source filter 
 ;   Upload options: custom upload port, speed and extra flags 
 ;   Library options: dependencies, extra library storages 
 ;   Advanced options: extra scripting 
 ; 
 ; Please visit documentation for the other options and examples 
 ; https://docs.platformio.org/page/projectconf.html 
  
 [env:esp32-s3] 
 platform = espressif32 @^6.3.2 
 board = esp32-s3-devkitc-1-n16r8v 
 framework = espidf 
 monitor_speed=115200 
 monitor_filters=esp32_exception_decoder 
 build_type = debug 
 board_build.partitions=./partition.csv 
 board_build.esp-idf.preserve_source_file_extension= yes`

coderforlife, 2025-08-17:
See #112 for another fix (the fix that would be the long term fix).