espressif/esp-dsp #75: esp32s3 fft4real hangs if fft4r related codes are removed.

wuyuanyi135, 2023-09-04:
## Environment

- Development Kit:      
- Kit version (for WroverKit/PicoKit/DevKitC): 
- Module or chip used:  ESP32-S3-WROOM-1-N4
- IDF version (run ``git describe --tags`` to find it): v5.2-dev-2383-g82cceabc6e
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp-12.2.0_20230208) 12.2.0
- Operating System:     Windows
- Power Supply:         USB

## Problem Description
I modified the fft for real example to only use fft2r functions. The program hangs.

### Expected Behavior
fft2r should work

### Actual Behavior
after remove fft4r related code, the code hangs.

### Steps to repropduce

see the code below

### Code to reproduce this issue

```c
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "driver/spi_master.h"
#include "soc/gpio_struct.h"
#include "driver/gpio.h"
#include "driver/uart.h"
#include "soc/uart_struct.h"
#include <math.h>

#include "esp_dsp.h"

static const char *TAG = "main";

// This example shows how to use FFT from esp-dsp library

#define N_SAMPLES 2048 // Amount of real input samples
int N = N_SAMPLES;
// Input test array
__attribute__((aligned(16)))
float x1[N_SAMPLES];
// Window coefficients
__attribute__((aligned(16)))
float wind[N_SAMPLES];
// Pointers to result arrays
float* y1_cf = &x1[0];

extern "C" void app_main()
{
  esp_err_t ret;
  ESP_LOGI(TAG, "Start Example.");
  ret = dsps_fft2r_init_fc32(NULL, N>>1);
  if (ret  != ESP_OK)
  {
    ESP_LOGE(TAG, "Not possible to initialize FFT2R. Error = %i", ret);
    return;
  }

  // Generate hann window
  dsps_wind_hann_f32(wind, N);
  // Generate input signal for x1 A=1 , F=0.1
  dsps_tone_gen_f32(x1, N, 1.0, 0.16,  0);

  // Convert two input vectors to one complex vector
  for (int i=0 ; i< N ; i++)
  {
    x1[i] = x1[i] * wind[i];
  }
  // FFT Radix-2
  unsigned int start_r2 = dsp_get_cpu_cycle_count();
  dsps_fft2r_fc32(x1, N>>1);
  // Bit reverse
  dsps_bit_rev2r_fc32(x1, N>>1);
  // Convert one complex vector with length N/2 to one real spectrum vector with length N/2
  dsps_cplx2real_fc32(x1, N>>1);
  unsigned int end_r2 = dsp_get_cpu_cycle_count();

  for (int i = 0 ; i < N/2 ; i++) {
    x1[i] = 10 * log10f((x1[i * 2 + 0] * x1[i * 2 + 0] + x1[i * 2 + 1] * x1[i * 2 + 1] + 0.0000001)/N);
  }

  // Show power spectrum in 64x10 window from -100 to 0 dB from 0..N/4 samples
  ESP_LOGW(TAG, "Signal x1");
  dsps_view(x1, N/2, 64, 10,  -60, 40, '|');
  ESP_LOGI(TAG, "FFT Radix 2 for %i complex points take %i cycles", N/2, end_r2 - start_r2);

  ESP_LOGI(TAG, "End Example.");
}
```

## Debug Logs

```
I (27) boot: ESP-IDF v5.2-dev-2383-g82cceabc6e 2nd stage bootloader
I (27) boot: compile time Sep  4 2023 10:24:09
I (27) boot: Multicore bootloader
I (32) boot: chip revision: v0.1
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : DIO
I (45) boot.esp32s3: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=202d4h (131796) map
I (125) esp_image: segment 1: paddr=000302fc vaddr=3fc90100 size=027e8h ( 10216) load
I (127) esp_image: segment 2: paddr=00032aec vaddr=40374000 size=0c048h ( 49224) load
I (141) esp_image: segment 3: paddr=0003eb3c vaddr=00000000 size=014dch (  5340)
I (143) esp_image: segment 4: paddr=00040020 vaddr=42000020 size=48ed8h (298712) map
I (206) boot: Loaded app from partition at offset 0x10000
I (206) boot: Disabling RNG early entropy source...
I (217) cpu_start: Multicore app
I (217) cpu_start: Pro cpu up.
I (218) cpu_start: Starting app cpu, entry point is 0x403750cc
0x403750cc: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-master/components/esp_system/port/cpu_start.c:170

I (0) cpu_start: App cpu up.
I (235) cpu_start: Pro cpu start user code
I (236) cpu_start: cpu freq: 240000000 Hz
I (236) cpu_start: Application information:
I (239) cpu_start: Project name:     aaar_wifi_tempvib
I (244) cpu_start: App version:      1
I (249) cpu_start: Compile time:     Sep  4 2023 10:24:00
I (255) cpu_start: ELF file SHA256:  cf4ffb33c...
I (260) cpu_start: ESP-IDF:          v5.2-dev-2383-g82cceabc6e
I (267) cpu_start: Min chip rev:     v0.0
I (271) cpu_start: Max chip rev:     v0.99
I (276) cpu_start: Chip rev:         v0.1
I (281) heap_init: Initializing. RAM available for dynamic allocation:
I (288) heap_init: At 3FC987D8 len 00050F38 (323 KiB): DRAM
I (294) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (301) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (307) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (314) spi_flash: detected chip: generic
I (318) spi_flash: flash io: dio
W (322) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (337) sleep: Configure to isolate all GPIO pins in sleep state
I (342) sleep: Enable automatic switching of GPIO sleep configuration
I (349) app_start: Starting scheduler on CPU0
I (354) app_start: Starting scheduler on CPU1
I (354) main_task: Started on CPU0
I (364) main_task: Calling app_main()
I (364) main: Start Example.

***** NO MORE LINES *****
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

peter-marcisovsky, 2023-09-04:
Hello @wuyuanyi135 could you please leave the `dsps_fft4r_init_fc32` in your code, and see if you could run it?

wuyuanyi135, 2023-09-05:
> Hello @wuyuanyi135 could you please leave the `dsps_fft4r_init_fc32` in your code, and see if you could run it?

Yes. with `dsps_fft4r_init_fc32` the example runs. I found `dsps_cplx2real_fc32` is actually a part of fft4r. I guess this is the issue why uninitialized fft4r caused problem.

wuyuanyi135, 2023-09-07:
@peter-marcisovsky sorry to bother you. Does `dsps_fft4r_init_fc32` affecting radix 2 APIs rings any alarm to you?

wuyuanyi135, 2023-09-22:
Anyone could provide some insights?