espressif/esp-dsp #76: SOLVED: TRIAC delayed-trigger compensation routine

f4lc0n-asm, 2023-11-12:
This is a highly optimized **TRIAC delayed-trigger compensation routine**.
It compensates the TRIAC trigger time so that the power delivered to resistive load is changing linearly when the original TRIAC trigger time is also changing linearly.

**Input:**  Relative delay of TRIAC trigger (0: start of half-sine wave up to 1: end of half-sine wave)
**Output:** Linearized relative delay (0 up to 1, 0 when input < 0, 1 when input > 1, NaN when input is NaN)
**Relative error of power delivered to resistive load:** < 1e-5

**Cycles:** 15-187 (includes CALL8, close to upper bound most of the time unless special input cases mentioned above)

**Comment:**
The power delivered with TRIAC to resistive load is simly **Integrate[Sin[x]^2, x]**, which yields **x/2 - 1/4 Sin[2 x]**.
But **InverseFunction[Function[x, x/2 - 1/4 Sin[2 x]]]** will get you nowhere :/
That is why I created this simple and most importantly **fast** routine!
Ring up your prodigy mathematician acquaintances - **maybe**, they will be able to devise an infinite series for the inverse function and conveniently truncate it when the required precision is met.

I am attaching the sources. Use 7-Zip to extract them from the 7-Zip archive. Feel free to incorporate them into this project. Let me know if you need me to tweak the ASM source.

Cheers!

f4lc0n

[dcomp_v1.0.zip](https://github.com/espressif/esp-dsp/files/13328698/dcomp_v1.0.zip)

TD-er, 2023-11-12:
Why in assembly?
Is this compatible with all ESP32 platforms?
It looks like it is a big lookup table, so I can imagine it is fast.
But still that doesn't explain why it has to be in assembly code.

Also I guess it would be nice to have a bit more explanation about the actual use case and what makes this a requirement for that use case.

f4lc0n-asm, 2023-11-12:
The ASM version is **2× faster** than C -Ofast!!! Since it extensively uses **Xtensa FPU**, it is compatible with **ESP32-S** and **ESP32-S3**. Is is suitable for cases when you want to **linearly** control the power delivered to the load connected to AC mains or any sinus power source in general.