espressif/esp-dsp #77: FIR Convolution algorithm error

Davidliujx, 2023-11-13:
FIR in the esp-dsp lib :  
esp_err_t dsps_fir_f32_ansi(fir_f32_t *fir, const float *input, float *output, int len)
{
    for (int i = 0 ; i < len ; i++) {
        float acc = 0;
        **int coeff_pos = 0;**                   //int coeff_pos = fir->N - 1;
        fir->delay[fir->pos] = input[i];
        fir->pos++;
        if (fir->pos >= fir->N) {
            fir->pos = 0;
        }
        for (int n = fir->pos; n < fir->N ; n++) {
            **acc += fir->coeffs[coeff_pos++] * fir->delay[n];**    //acc += fir->coeffs[coeff_pos--] * fir->delay[n];
        }
        for (int n = 0; n < fir->pos ; n++) {
            **acc += fir->coeffs[coeff_pos++] * fir->delay[n];**  //acc += fir->coeffs[coeff_pos--] * fir->delay[n];
        }
        output[i] = acc;
    }
    return ESP_OK;
}

dmitry1945, 2023-11-13:
Hi @Davidliujx,
We've changed the  `fir->coeffs[coeff_pos--]` to the ` fir->coeffs[coeff_pos++]` for speed optimization reason. The implementation In this way is just faster.
But, for this, the order of your coefficients should be reversed. 
I will add the comment to the header about coefficients order.
Thank you,
Dmitry

TD-er, 2023-11-13:
If you're really into speed optimizations, then you may also want to look at lines like these:
```c++
for (int n = fir->pos; n < fir->N ; n++) {
```

Using `++n` in a loop is faster.
Make as much as possible `const` in the invariant.

Thus: 
```c++
const int fir_N = fir->N;
for (int n = fir->pos; n < fir_N; ++n) {
```

dmitry1945, 2023-11-13:
@TD-er  yes, will add this to the list (it's not only FIR feature).

Thank you,
Dmitry

Davidliujx, 2023-11-14:
thanks for your all reply

closed 2023-11-14 by Davidliujx