espressif/esp-dsp #79: Mismatch between latest repository tag and latest component manager tag

herculanodavi, 2023-12-21:
Hello, I'd like to better understand the release process of this repository and its sync with component-manager. As stated in the title, the latest tag here is `1.4.0` while it is `1.4.10` in component-manager, which hints that there were a lot of fixes in between.

So, are we able to see which fixes were done? Is the code in `1.4.0` the latest? How can I see the `1.4.10` code?

Thanks for your support.

dmitry1945, 2023-12-21:
Hi @herculanodavi ,

Yes, we've miss the tag in repo. Will add it to the release process.

Now we have tag in  idf_component.yml, and this is the same as in component manager.

Thank you,
Dmitry

herculanodavi, 2024-03-28:
Hi Dmitri, thanks a lot but I really want to checkout the library code and clone it to my machine in the 1.4.10 commit, but I can't since the tag is not in GitHub. Would you be able to just create a tag on where the commit is, please?

dmitry1945, 2024-03-28:
Hi @herculanodavi,
Please take a look. The tag now is on v1.4.10.

closed 2024-03-31 by herculanodavi