espressif/esp-dsp #85: Compile `esp-dsp` for `IDF_TARGET=linux`

morgana2313, 2024-04-05:
### Describe the solution you'd like.

I would like to be able to compile `esp-dsp` for `IDF_TARGET=linux` so I can do (unit) tests on the host computer without an actual esp32. 

When I try to compile `example/basic_math` it fails:
```
basic_math $ IDF_TARGET=linux idf.py build
Executing action: all (aliases: build)
Running cmake in directory /Users/jasper/esp/esp-dsp/examples/basic_math/build
[...]
-- Building ESP-IDF components for target linux
[...]
In file included from esp/esp-dsp/modules/common/misc/aes3_tie_log.c:7:
In file included from esp/esp-dsp/modules/common/include/dsp_common.h:23:
esp/esp-idf/components/esp_hw_support/include/esp_cpu.h:129:17: error: call to undeclared function 'rv_utils_get_core_id'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
    return (int)rv_utils_get_core_id();
[...]
fatal error: too many errors emitted, stopping now [-ferror-limit=]
```

dmitry1945, 2025-04-16:
Hi @morgana2313 

the support for Linux added.

Thank you,
Dmitry

morgana2313, 2025-04-16:
> the support for Linux added.

Thank you Dmitry!  

From https://github.com/espressif/esp-dsp/pull/103#issuecomment-2809250728 I suspect this won't work on Mac with Apple silicon (AArch64)?

closed 2026-10-20 by dmitry1945