espressif/esp-dsp #90: P4 compile error

Jason2866, 2024-08-12:
### Answers checklist.

- [X] I have read the [ESP-DSP documentation](https://docs.espressif.com/projects/esp-dsp/en/latest/esp-dsp-library.html/) and the issue is not addressed there.
- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my ESP-DSP branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF 5.2 latest master

### Module or chip used.

Compile try for ESP32-P4

### DSP version.

latest master

### What is the expected behavior?

Code compiles

### What is the actual behavior?

compile error with `modules/dotprod/float/dsps_dotprode_f32_arp4.S:52: Error: unrecognized opcode `esp.lp.setup 0,a3,.dotprod_loop'`

### Steps to reproduce.

Just tring to compile fails

### Code to reproduce this issue.

```plain
Not needed
```

### Debug logs.

See CI run of Arduino Lib Builder https://github.com/Jason2866/esp32-arduino-lib-builder/actions/runs/10341546001/job/28623471052

dmitry1945, 2024-08-13:
Hi @Jason2866,
Thank you for feedback.
 
Could you try to switch to latest idf master branch?
Then ./install.sh and then . ./export.sh

Now only latest master has a toolchain to support esp32p4 instructions like `esp.lp.setup`

Please let me know the result.
Thank you.

Jason2866, 2024-08-13:
Understood, branch IDF 5.2 is not supported. Can't change to IDF master branch easily, many code changes needed. Will the needed changes backported to IDF 5.3.x and 5.2.x
If not, no drama will exclude DSP lib when compiling for P4

dmitry1945, 2024-08-13:
@Jason2866 I think it's a plans to make a backport.
Now, you can just take binutils from latest master, and use them.
Yes, I will add restriction to idf_component.yml 
Thank you.

dmitry1945, 2024-08-13:
@Jason2866 
Also, you can now disable optimization for esp-dsp in menuconfig, and you will be able to use the esp-dsp. For dloating point operations, you will not lose a lot of performance. And in future will fix the problem.

closed 2025-06-09 by Indastri