espressif/esp-dsp #93: ESP-DSP: DC-Offset and Nyquist frequency in output?

ruig25, 2024-09-05:
### Answers checklist.

- [X] I have read the [ESP-DSP documentation](https://docs.espressif.com/projects/esp-dsp/en/latest/esp-dsp-library.html/) and the issue is not addressed there.
- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my ESP-DSP branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi!

I suspect, that fft output contains dc offset at index=0 and Nyquist frequency at index=1 (like in ARM CMSIS DSP library for cortex).

I have accelerometer signal. I feed input signal (hann window applied to real values) interleaved by zeros to dsps_fft2r_fc32.

1) Bit reverse op applied to output (dsps_bit_rev_fc32).
2) dsps_cplx2reC_fc32 is also applied.

After calc of magnitude on output sqrtf(real^2 + imag^2) the first bin (index = 1) is always bigger than others.
If I set bin 0 and 1 to zero - I get the results that seems to be near to reality (bin index multiplied by freq resolution... ~ expected fundamental freq of my test vibration).

I am i right about index 0 and 1?

Thanks

ruig25, 2024-09-12:
[image]

dmitry1945, 2024-09-14:
Hi @ruig25 
Could you please share here the input data that you are converting? Just 512 samples?
Thank you.

ruig25, 2024-09-16:
Hello!

Sure.

**1st example**
Accelerometer ST lsm6dsl (+/-2g, Fs=1,66kHz, no filters).
FFT_SIZE=512.
Hann Windows size = FFT_SIZE
[image]

**Raw accelerometer 512 samples used to calc magnitude**
[xl_3d_raw.json](https://github.com/user-attachments/files/17009275/xl_3d_raw.json)

ruig25, 2024-09-16:
**2nd example**

Source: 3D magnet field sensor.
Fs=800Hz
No filters.

The same routine as above but RAW values converted to milli Tesla before calculating magnitude.

In both cases first two FFT bins (0 and 1) contain high values.

[b_3d_mt.json](https://github.com/user-attachments/files/17009966/b_3d_mt.json)

I will try to prepare Octave snippet to verify fft... may my fault somewhere

dmitry1945, 2024-09-19:
Hi @ruig25 

You have a positive bin[0] and this bin is bigger then others for tree reasons. 
1. All samples that you have are positive. It means, that they have DC value inside.
2. Even if you will have negative values, you are using sqrt(A*A + B*B + C*C) function. It also provide you only positive values, and it means, it has DC value inside.
3. Bin[1] has high value because you are using window function. Even if you don't have it, the bin[1] will be max -13 dB smaller then bin[0].

Regards,
Dmitry.