espressif/esp-dsp #94: cplx2real function can only be used with fft4r initialized, and the example creates confusion

alexbarcelo, 2024-09-24:
https://github.com/espressif/esp-dsp/blob/10d74183ee8e55113a964dcf92c0d9ed639cae03/examples/fft4real/main/dsps_fft4real_main.c#L85

I was trying to follow the example with a codebase that uses not fft4r but only fft2r functions (which are showcased in this example, I understand). I was getting segfault errors and after gazillion checks in the documentation I realized that cplx2real documentation says that fft4r needs to be initialized. In the example may be ok because it is showcasing both, but in a regular scenario it is not ok.

Assuming that this is indeed the problem (my code started working once I changed to cplx2reC_fc32 function, but I have not validated the numerical aspect, so I may have more bugs, and I may have misunderstood the documentation once again), the example should use the proper function to showcase properly which functions to use in which scenarios.

dmitry1945, 2024-09-25:
Hi @alexbarcelo 

The  dsps_cplx2real_fc32 need initialization for FFT4R. You have to call `dsps_fft2r_init_fc32` and  `dsps_fft4r_init_fc32`.
Did you call `dsps_fft4r_init_fc32`?

alexbarcelo, 2024-09-25:
Of course everything is due to my ignorance and incompetence. I was assuming that cplx2real and cplx2reC were "similar". Maybe they are not?

I assumed that the example showcases two different "flows": the fft2r_* way and the fft4r_* way. But from your comment, when using dsps_fft2r mehods I need to do the init also for fft4r. Is this the case for dsps_cplx2real_sc16_ansi? Documentation does not mention 4r in that, there is no 4r init, so I was confused.

dmitry1945, 2024-09-25:
@alexbarcelo 
The use a table and additional memory, that's why we have moved the cplx2real  to the fft4r init function.
I will check the functions later.

stccff, 2025-05-08:
I'm very confused!!! I was getting panic as well.
The fft libarary and examples are terrible.

stccff, 2025-05-08:
> [esp-dsp/examples/fft4real/main/dsps_fft4real_main.c](https://github.com/espressif/esp-dsp/blob/10d74183ee8e55113a964dcf92c0d9ed639cae03/examples/fft4real/main/dsps_fft4real_main.c#L85)
> 
> Line 85 in [10d7418](/espressif/esp-dsp/commit/10d74183ee8e55113a964dcf92c0d9ed639cae03)
> 
>  dsps_cplx2real_fc32(x1, N >> 1); 
> I was trying to follow the example with a codebase that uses not fft4r but only fft2r functions (which are showcased in this example, I understand). I was getting segfault errors and after gazillion checks in the documentation I realized that cplx2real documentation says that fft4r needs to be initialized. In the example may be ok because it is showcasing both, but in a regular scenario it is not ok.
> 
> Assuming that this is indeed the problem (my code started working once I changed to cplx2reC_fc32 function, but I have not validated the numerical aspect, so I may have more bugs, and I may have misunderstood the documentation once again), the example should use the proper function to showcase properly which functions to use in which scenario
where is the docs of fft?

alexbarcelo, 2025-05-08:
@stccff the docs are in the espressif's docs website:

https://docs.espressif.com/projects/esp-dsp/en/latest/esp32/esp-dsp-apis.html#fft

(And also the examples and their explanation)

yousseffaouzi, 2026-10-20:
I checked the current `master` implementation (`3c8ac0fdfec83740b783e200862c8d0c056de0ad`) and I think there is a narrow fix we can make here without changing the FFT API.

The FFT4 dependency itself appears intentional: `dsps_cplx2real_fc32()` uses the FFT4 twiddle table, so even when it follows the radix-2 FFT flow, `dsps_fft4r_init_fc32()` is still required. Also, `dsps_cplx2reC_fc32()` is not a drop-in replacement; it serves a different packed-two-real-signals use case.

One concrete inconsistency I found is that the ANSI implementation of `dsps_cplx2real_fc32_ansi_()` checks `dsps_fft4r_initialized` and returns `ESP_ERR_DSP_UNINITIALIZED`, while the optimized `dsps_cplx2real_fc32_ae32_()` currently has no equivalent guard before using the FFT4 table. That can turn a missing initialization into a crash/panic instead of a clean error on optimized targets.

If this direction is acceptable, I can prepare a small PR that:

- adds the missing initialization guard to the AE32 path so ANSI and optimized behavior match;
- adds regression coverage for the uninitialized case;
- clarifies the `dsps_cplx2real_fc32()` documentation that FFT4 initialization is required even after a radix-2 FFT because the post-processing uses the FFT4 table;
- improves the `fft4real` example comments/error handling so the reason for initializing both FFT2 and FFT4 is explicit.

I would keep any larger redesign of the real-FFT post-processing table/API separate.

Does that scope make sense?