espressif/esp-dsp #106: Title: ESP-DSP Support for ESP32C5

coyote0504, 2025-04-11:
### Answers checklist.

- [x] I have read the [ESP-DSP documentation](https://docs.espressif.com/projects/esp-dsp/en/latest/esp-dsp-library.html/) and the issue is not addressed there.
- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my ESP-DSP branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Description: Hi Espressif team, I am wondering whether ESP-DSP is compatible with the ESP32C5 chip. The official documentation and repository mention support for ESP32, ESP32-S3, and ESP32P4, but there is no mention of ESP32C5. Is there any planned support for ESP32C5 in the future? If not, is there a recommended approach to use ESP-DSP functionalities on ESP32C5?

Thank you!

dmitry1945, 2025-04-14:
Hi @coyote0504 

You can use the esp-dsp with esp32c5, it's not a problem.

Dmitry

tookitogo, 2026-10-02:
> The official documentation and repository mention support for ESP32, ESP32-S3, and ESP32P4

Well, actually it mentions says it has _optimized functions_ for those models, not that it doesn’t work on others. (That is, the others just don’t have specially-optimized functions.)