espressif/esp-dsp #107: 使用ESP-SR库的时候报错。

ZhouKes, 2025-04-17:
### Answers checklist.

- [x] I have read the [ESP-DSP documentation](https://docs.espressif.com/projects/esp-dsp/en/latest/esp-dsp-library.html/) and the issue is not addressed there.
- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my ESP-DSP branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Module or chip used.

ESP32S3

### DSP version.

V1.6.0

### What is the expected behavior?

之前使用esp-sr库可正常编译

### What is the actual behavior?

esp-dsp库昨天自动更新后，编译报错

### Steps to reproduce.

1，新建项目，添加esp-sr组件依赖
2，在项目中使用esp-sr库
3，编译
...

### Code to reproduce this issue.

```plain
/home/sunxiangyu/workspace/esp_sr_lib/components/c_speech_features/esp_speech_features/esp_speech_features.c:294:(.text.esp_fftr+0x31): undefined reference to `dsps_fft4r_fc32_ae32_'
```

### Debug logs.

/home/sunxiangyu/workspace/esp_sr_lib/components/c_speech_features/esp_speech_features/esp_speech_features.c:294:(.text.esp_fftr+0x31): undefined reference to `dsps_fft4r_fc32_ae32_'

ZhouKes, 2025-04-17:
手动修改esp-sr库对esp-dsp库的版本需求，退回到之前的版本，编译通过。

ZhouKes, 2025-04-17:
@dmitry1945  esp-dsp was added by the component esp-sr, I can find it in managed_components

ZhouKes, 2025-04-17:
@dmitry1945 ,if I modify the idf_componet.yml in esp-sr,

dmitry1945, 2025-04-17:
@ZhouKes OK. Thank you. I will check.

Dmitry

dmitry1945, 2025-04-17:
@ZhouKes  one more question. For wich cpu are compiling? For esp32s3? Or for esp32?

dmitry1945, 2025-04-17:
@ZhouKes  the problem is simple.
I last version of esp-dsp the function dsps_fft4r_fc32_ae32_ was splitted to two functions: dsps_fft4r_fc32_ae32_ and dsps_fft4r_fc32_aes3_. When you compile your libraty (esp_sr in our case), you don't change the interface, and it will work. But, for you, you are using old esp_sr and new esp-dsp that has no dsps_fft4r_fc32_ae32_ for esp32s3, and you have an error.
I will fix it now.

Thank you for your issue.

Dmitry

dmitry1945, 2025-04-18:
@ZhouKes the lates esp-dsp should work with esp_sr. Please check.

Dmitry

ZhouKes, 2025-04-18:
@dmitry1945 Thank you!

closed 2025-04-28 by dmitry1945