espressif/esp-dsp #108: Strange behavior of dsps_fft2r_sc16_arp4() on ESP32P4

morihirotc, 2025-04-24:
### Answers checklist.

- [x] I have read the [ESP-DSP documentation](https://docs.espressif.com/projects/esp-dsp/en/latest/esp-dsp-library.html/) and the issue is not addressed there.
- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my ESP-DSP branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Module or chip used.

ESP32P4

### DSP version.

v1.6.1

### What is the expected behavior?

Obtain same results as dsps_fft2r_sc16_ansi() and buffer overrun does not caused.

example output by dsps_fft2r_sc16_ansi().
[image]

### What is the actual behavior?

Unexpected results obtained and cause buffer overrun at *data by 32Bytes. And the results are little bit noisy.

example output by dsps_fft2r_sc16_arp4().
[image]

### Steps to reproduce.

1.  Implement the sc16 version based on the fc32 example code and modified a little
2.  Flash the program to MCU and Run
3.  The issue is caused.

[sdkconfig.txt](https://github.com/user-attachments/files/19880316/sdkconfig.txt)

### Code to reproduce this issue.

```plain
All the code I used to reproduce this bug is here.
https://gist.github.com/morihirotc/4e45fd8eec18501ac46829aa7092b077
```

### Debug logs.

All serial output is here.
[log.txt](https://github.com/user-attachments/files/19880360/log.txt)

### More Information.

This bug is quite elusive and has proven very difficult to consistently reproduce. For example, simply changing the function used to generate input data or adding another function can cause the issue to appear or disappear.

dmitry1945, 2025-04-25:
Hi @morihirotc 
I think the reason for the error is that you use my_cplx2reC_sc16(). The function shift second spectrum to one bin. It make the error.

The noise, that you see between ansi and p4 implementation I can explain. 
The ansi implementation use rounding between FFT stages, and P4 not.
The P4 implementation is around 10 times faster.  The noise here is a compromise between speed and noise.
In version 2.0 we will have FFT that will little slower then now, but will have no noise, like ansi.

Dmitry

morihirotc, 2025-04-28:
Hi @dmitry1945 
Thanks to reply.
No, my_cplx2reC_sc16() does not cause the issues. It is a modified version of dsps_cplx2reC_sc16(), and I have confirmed that it produces correct results when combined with dsps_fft2r_sc16_ansi(). I am using my_cplx2reC_sc16() because the results of dsps_cplx2reC_sc16() differed from dsps_cplx2reC_fc32(), and I have made adjustments accordingly.
Additionally, I have tried switching between float/int versions and changing their usage, but dsps_fft2r_sc16_arp4() still does not yield the desired results.
Since a buffer overflow seems to be occurring in dsps_fft2r_sc16_arp4(), I suspect that this function is the cause of the issue.
However, I am not entirely familiar with the library's usage, so please let me know if there are any mistakes.

And, I understand the issue regarding the noise. Thank you .

dmitry1945, 2025-04-28:
Hi @morihirotc ,

Finally, I have understood and found the problem.
Will publish the fix soon.

Thank you very much!

dmitry1945, 2025-04-28:
@morihirotc 
The problem should be fixed now.
please update your version to the latest one.

Dmitry

morihirotc, 2025-04-30:
Hi, @dmitry1945 
I’ve updated to v1.6.2, and the code now seems to be running correctly. Thank you so much for your help.

closed 2025-04-30 by dmitry1945