espressif/esp-dsp #113: ESP32P4 Compile error

iothub01, 2025-09-02:
### Answers checklist.

- [x] I have read the [ESP-DSP documentation](https://docs.espressif.com/projects/esp-dsp/en/latest/esp-dsp-library.html/) and the issue is not addressed there.
- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my ESP-DSP branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Module or chip used.

ESP32P4

### DSP version.

v1.6.0

### What is the expected behavior?

Work Fine

### What is the actual behavior?

Whenever i am trying to use AEC feature in my webrtc code , i am getting compilation error - 
managed_components/espressif__esp-sr/lib/esp32p4/libc_speech_features.a(esp_speech_features.c.obj): in function `esp_mel_dotprod_f32':
/sunxiangyu/workspace/esp_sr_lib/components/c_speech_features/esp_speech_features/esp_speech_features.c:209:(.text.esp_mel_dotprod_step+0x58): undefined reference to `dsps_dotprod_f32_arp4'
undefined reference to `dsps_dotprod_f32_arp4'
undefined reference to `dsps_fft2r_fc32_arp4_'

### Steps to reproduce.

I am using doorbell example of https://github.com/espressif/esp-webrtc-solution and updating build_capture_system function inside media_sys.c with following code 

`static int build_capture_system(void)
{
    capture_sys.vid_src = create_video_source();
    RET_ON_NULL(capture_sys.vid_src, -1);

    // Use AEC audio source with proper configuration
    esp_capture_audio_aec_src_cfg_t aec_cfg = {
        .mic_layout = "M",           // Single microphone layout
        .record_handle = get_record_handle(),
        .channel = 1,                // Mono audio
        .channel_mask = 0x1,         // Process first channel
    };
    
    capture_sys.aud_src = esp_capture_new_audio_aec_src(&aec_cfg);
    RET_ON_NULL(capture_sys.aud_src, -1);

    // Create capture system
    esp_capture_cfg_t cfg = {
        .sync_mode = ESP_CAPTURE_SYNC_MODE_AUDIO,
        .audio_src = capture_sys.aud_src,
        .video_src = capture_sys.vid_src,
    };
    esp_capture_open(&cfg, &capture_sys.capture_handle);
    
    return 0;
}`

### Code to reproduce this issue.

```plain
static int build_capture_system(void)
{
    capture_sys.vid_src = create_video_source();
    RET_ON_NULL(capture_sys.vid_src, -1);

    // Use AEC audio source with proper configuration
    esp_capture_audio_aec_src_cfg_t aec_cfg = {
        .mic_layout = "M",           // Single microphone layout
        .record_handle = get_record_handle(),
        .channel = 1,                // Mono audio
        .channel_mask = 0x1,         // Process first channel
    };
    
    capture_sys.aud_src = esp_capture_new_audio_aec_src(&aec_cfg);
    RET_ON_NULL(capture_sys.aud_src, -1);

    // Create capture system
    esp_capture_cfg_t cfg = {
        .sync_mode = ESP_CAPTURE_SYNC_MODE_AUDIO,
        .audio_src = capture_sys.aud_src,
        .video_src = capture_sys.vid_src,
    };
    esp_capture_open(&cfg, &capture_sys.capture_handle);
    
    return 0;
}
```

### Debug logs.

managed_components/espressif__esp-sr/lib/esp32p4/libc_speech_features.a(esp_speech_features.c.obj): in function `esp_mel_dotprod_f32':
/sunxiangyu/workspace/esp_sr_lib/components/c_speech_features/esp_speech_features/esp_speech_features.c:209:(.text.esp_mel_dotprod_step+0x58): undefined reference to `dsps_dotprod_f32_arp4'
undefined reference to `dsps_dotprod_f32_arp4'
undefined reference to `dsps_fft2r_fc32_arp4_'

wezhunter, 2025-09-15:
I've discovered this is a config issue as part of my project migration from initial release esp-sr + esp-dsp on the ESP32-P4.

sdconfig needs to be set as -
CONFIG_DSP_OPTIMIZATIONS_SUPPORTED=y
# CONFIG_DSP_ANSI is not set
CONFIG_DSP_OPTIMIZED=y
CONFIG_DSP_OPTIMIZATION=1

Otherwise `dsps_dotprod_s16_arp4_enabled = 0` and `dsps_dotprod_f32` is not included in `dsps_dotprod.h` header as part of the esp-dsp component.

This fixes all the above include errors.

Might be worthwhile adding a note on the ESP-SR/DSP Readme page regarding the ESP-DSP config option and/or adding a warning in the include header if ESP-DSP optimisation is set to ANSI and SR is being used on the P4? 

I could not find anything regarding this specific what appears to be relatively new menu option anywhere AFAIK.

Thanks

sun-xiangyu, 2025-11-03:
> I've discovered this is a config issue as part of my project migration from initial release esp-sr + esp-dsp on the ESP32-P4.
> 
> sdconfig needs to be set as - CONFIG_DSP_OPTIMIZATIONS_SUPPORTED=y #CONFIG_DSP_ANSI is not set CONFIG_DSP_OPTIMIZED=y CONFIG_DSP_OPTIMIZATION=1
> 
> Otherwise `dsps_dotprod_s16_arp4_enabled = 0` and `dsps_dotprod_f32` is not included in `dsps_dotprod.h` header as part of the esp-dsp component.
> 
> This fixes all the above linker errors.
> 
> It might be worthwhile adding a note on the ESP-SR/DSP Readme page regarding the ESP-DSP config option and/or adding a warning in the include header if ESP-DSP optimisation is set to ANSI and SR is being used on the P4?
> 
> I could not find anything mentioned anywhere regarding this specific, and what appears to be relatively new, menu option (AFAIK).
> 
> Thanks

Thanks for your suggestion. I will update esp-sr.